nickg/nvc #100: In accurate REAL'IMAGE

diogratia, 2015-01-19:
entity tell_real_range is
    end entity;
    architecture foo of tell_real_range is
    begin
        assert FALSE
        report "real range from " & real'image(real'low) & " to " 
                                  & real'image(real'high)
        severity NOTE;
    end architecture;

Produces:

    nvc -r tell_real_range
    ** Note: 0ms+0: Assertion Note: real range from -1.79769e+308 to 1.79769e+308
    	Process :tell_real_range:line_6
    	File tell_real_range.vhdl, Line 6

Where the actual values should be +/- 1.7976931348623157e308.

From 14.1 Predefined attributes for attribute 'IMAGE:

> ... If T is a numeric type or subtype, the result is expressed as the decimal representation of the parameter value without underlines or leading or trailing zeros (except as necessary to form the image of a legal literal with the proper value); moreover, an exponent may (but is not required to) be present and the language does not define under what conditions it is or is not present. If the exponent is present, the "e" is expressed as a lowercase character.  ... 
>  If T is a floating point type or type, the number of digits to the right of the decimal point corresponds to the standard form generated when the DIGITS parameter to TextIO. Write for type REAL is set to 0 (see 14.3 ). 

(Also in 16.2 in -2008).

In 14.3:
> The definitions of the string representation of the value for each data type are as follows:   
 ...  
-- The representation of both INTEGER and REAL values is that of a decimal literal (see 13.4.1 ), with the addition of an optional leading sign. The sign is never written if the value is nonnegative, but it is accepted during a read even if the value is nonnegative. No spaces can occur between the sign and the remainder of the value. The decimal point is absent in the case of an INTEGER literal and present in the case of a REAL literal. An exponent may optionally be present; moreover, the language does not define under what conditions it is or is not present. However, if the exponent is present, the "e" is written as a lowercase character. Leading and trailing zeroes are written as necessary to meet the requirements of the FIELD and DIGITS parameters, and they are accepted during a read.  
...  
For each predefined data type there is one WRITE procedure declared in package TEXTIO. Each of these has at least two parameters: L, the line to which to write; and VALUE, the value to be written. The additional parameters JUSTIFIED, FIELD, DIGITS, and UNIT control the formatting of output data. Each write operation appends data to a line formatted within a field that is at least as long as required to represent the data value. Parameter FIELD specifies the desired field width. Since the actual field width will always beat least large enough to hold the string representation of the data value, the default value 0 for the FIELD parameter has the effect of causing the data value to be written out in a field of exactly the right width (i.e., no leading or trailing spaces). Parameter JUSTIFIED specifies whether values are to be right- or left-justified within the field; the default is right-justified. If the FIELD parameter describes a field width larger than the number of characters necessary for a given value, blanks are used to fill the remaining characters in the field.  
  
> Parameter DIGITS specifies how many digits to the right of the decimal point are to be output when writing a real number; the default value 0 indicates that the number should be output in standard form, consisting of a normalized mantissa plus exponent (e.g. 1.079236E-23). If DIGITS is nonzero, then the real number is output as an integer part followed by '.' followed by the fractional part, using the specified number of digits (e.g., 3.14159 ).   

Where DIGITS = 0 as an argument to write does not give permission to use a number of digits less that the representation of the real number.

(And the same holds true in 16.4 in -2008).

hiyuh, 2015-02-03:
note, `_image()` of src/rt/rtkern.c uses `double` and `"%g"` for `T_REAL` via `int64_t`.

nickg, 2015-02-03:
I think we ought to be able to set the precision argument to "%g" to the maximum representable digits

nickg, 2015-02-03:
Should be fixed by the above. The + 3 comes from here:

http://stackoverflow.com/a/19897395

closed 2015-02-03 by nickg