nickg/nvc #101: No error when arthmetic operator and mathematical value mismatch.

diogratia, 2015-01-25:
entity example is
    end entity;

    architecture SIGN of EXAMPLE is
        signal TRIGGER, RESULT: integer := 0; 
        signal signal1: integer :=1;
        signal signal2: integer :=2;
        signal signal3: integer :=3;
    begin
        process (signal1, signal2, signal3)
        begin
            -- wait on TRIGGER;
            signal1 <= signal2;
            signal2 <= signal1 + signal3;
            signal3 <= signal2;
            RESULT <= signal1 + signal2 + signal3;
        end process;
    monitor:
        process(RESULT)
        begin
            report "RESULT = " & integer'image(RESULT);
        end process;
    end SIGN;

> ...
>** Note: 0ms+55: Report Note: RESULT = 805306368
>    Process :example:monitor
>    File sign.vhdl, Line 21
>** Note: 0ms+56: Report Note: RESULT = 1073741824
>    Process :example:monitor
>    File sign.vhdl, Line 21
> ** Note: 0ms+57: Report Note: RESULT = 1610612736
>    Process :example:monitor
>    File sign.vhdl, Line 21
>** Note: 0ms+58: Report Note: RESULT = -2147483648
>    Process :example:monitor
>    File sign.vhdl, Line 21
>** Note: 0ms+59: Report Note: RESULT = -1073741824
>    Process :example:monitor
>    File sign.vhdl, Line 21
>** Note: 0ms+60: Report Note: RESULT = 0
>    Process :example:monitor
>    File sign.vhdl, Line 21
>** Note: 0ms+61: Report Note: RESULT = -2147483648
>    Process :example:monitor
>    File sign.vhdl, Line 21
>** Note: 0ms+62: Report Note: RESULT = 0
>    Process :example:monitor
>    File sign.vhdl, Line 21

Where you can see the integer arithmetic rolled over.

IEEE Std 1076-2008, 5.2.3 Integer types, 5.2.3.1 General, paragraph 7/ -1993,
3.1.2 Integer types, paragraph 7 (note 9.2 reference is 7.2 in -1993):

>The same arithmetic operators are predefined for all integer types (see 9.2).
>It is an error if the execution of such an operation (in particular, an
>implicit conversion) cannot deliver the correct result (that is, if the value
>corresponding to the mathematical result is not a value of the integer type).

Instead of rolling over the results should have generated an error.

Detecting the mathematical/integer type value mismatch can either be by
performing infinite precision arithmetic or detecting incorrect sign (which
is what we tend to do in microprocessor design).  VHDL does not require
infinite precision.

hiyuh, 2015-02-03:
note, `cgen_fcall()` of `src/cgen.c` registers built-in `"add"` by `LLVMBuildAdd()`,
after that it returns w/ `cgen_narrow()` registers implicit `"narrow"` cast by `LLVMBuildIntCast()` to target type if required.

nickg, 2015-02-03:
Yes that's probably the place where such a check ought to go, but I'm in the middle of rewriting the code generator at the moment (see the new-codegen branch) so it's probably not worth fixing at the moment.

nickg, 2022-04-22:
This is fixed now:

```
** Note: 0ms+55: Report Note: RESULT = 805306368
        File /home/nick/nvc/test/regress/issue101.vhd, Line 21
** Note: 0ms+56: Report Note: RESULT = 1073741824
        File /home/nick/nvc/test/regress/issue101.vhd, Line 21
** Fatal: result of 1610612736 + 536870912 cannot be represented as INTEGER
        File /home/nick/nvc/test/regress/issue101.vhd, Line 16
    |
 16 |         RESULT <= signal1 + signal2 + signal3;
    |                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^
```

closed 2022-04-22 by nickg