nickg/nvc #102: declaration component not found in package unit

ikwzm, 2015-01-31:
```vhdl
-- file: a.vhd
library ieee;
use     ieee.std_logic_1164.all;
package COMPONENTS is
    component DUMMY_MODULE
        port (I : in std_logic; O : out std_logic);
    end component;
end package;

library ieee;
use     ieee.std_logic_1164.all;
use     WORK.COMPONENTS.DUMMY_MODULE;
entity  DUMMY_TOP is
    port (I : in std_logic; O : out std_logic);
end entity;
architecture RTL of DUMMY_TOP is
begin
    U: DUMMY_MODULE port map(I=>I, O=>O);
end architecture;
```
    ikwzm@vm-ubuntu:~/work/nvc_test$ nvc -a a.vhd
    ** Error: declaration DUMMY_MODULE not found in unit WORK.COMPONENTS
    	File a.vhd, Line 11
        use     WORK.COMPONENTS.DUMMY_MODULE;
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

hiyuh, 2015-02-03:
maybe, `scope_import_unit()` or `sem_check_use_clause()` are broken?

ikwzm, 2015-02-11:
scope_import_unit() is calling sem_declare(), but its' this in sem_declare().
````
static bool sem_declare(tree_t decl, bool add_predefined)
{
   if (tree_kind(decl) == T_COMPONENT)
       return true;
````
When sem_declare() was changed this, it didn't occur any more.
````
static bool sem_declare(tree_t decl, bool add_predefined)
{
   if (tree_kind(decl) == T_COMPONENT)
       return scope_insert(decl);

````
My change does not have the correct whether confidence....

nickg, 2015-02-11:
Yes I believe that's the correct fix. I added a test case too.

closed 2015-02-11 by nickg