nickg/nvc #103: The access type does not have a range attribute.

ikwzm, 2015-02-14:
````
-- file: a.vhd
use std.textio.all;
entity access_attribute_range is
end entity;
architecture test of access_attribute_range is
    procedure puts(TAG:inout LINE; MSG:in STRING) is
        variable text_line   : LINE;
    begin
        write(text_line, TAG(TAG'range), RIGHT,  10);
        write(text_line, string'(" "));
        write(text_line, MSG(MSG'range));
        writeline(OUTPUT, text_line);
    end procedure;
begin
    main: process
        variable tag  : line;
    begin
        write(tag , string'(">>"));
        puts(tag, "test");
        report "done." severity FAILURE;
        wait;
    end process;
end test;
````
````
ikwzm@vm-ubuntu:~/work/nvc_test$ nvc -a a.vhd
** Error: object RANGE does not have a range
	File a.vhd, Line 8
            write(text_line, TAG(TAG'range), RIGHT,  10);
                                 ^^^^^^^^^
````

nickg, 2015-03-01:
This error is fixed in the master branch however your example causes a crash at runtime. This is fixed in the new-codegen branch so I'll close this when I merge that branch back in.

nickg, 2015-04-25:
This is fixed now and produces the expected output.

closed 2015-04-25 by nickg