nickg/nvc #104: Another expression cannot be folded to an integer constant

swisssam, 2015-02-20:
Hi,
I wanted to check how nvc compares to ghdl on my design. However I could not fully elaborate my design.
This gives a "**Fatal: expression cannot be folded to an integer constant" at line 28 during elaboration on the latest version:
```vhdl
LIBRARY ieee;
USE ieee.std_logic_1164.ALL;
USE ieee.numeric_std.ALL;

ENTITY dummy IS PORT(
    i : IN std_logic;
    o : OUT std_logic );
END ENTITY dummy;

ARCHITECTURE arch OF dummy IS
BEGIN
    o <= i;
END ARCHITECTURE arch;

LIBRARY ieee;
USE ieee.std_logic_1164.ALL;
USE ieee.numeric_std.ALL;

ENTITY nvc_tb IS
END ENTITY nvc_tb;

ARCHITECTURE arch OF nvc_tb IS
    SIGNAL v : std_logic_vector(4 DOWNTO 0);
BEGIN
    fold : FOR i IN 0 TO 3 GENERATE
    BEGIN
        dummy : ENTITY work.dummy PORT MAP(
            i => v(i+1),
            o => v(i)
        );
    END GENERATE fold;
END ARCHITECTURE arch;
```

nickg, 2015-03-01:
Should work now

closed 2015-03-01 by nickg

swisssam, 2015-03-16:
Yes, this issue is solved, thank you.