nickg/nvc #105: relax "case choice must be locally static"

andreygursky, 2015-03-13:
Hi Nick,

please look at the following minimal example. Such code can be compiled and synthesized since 10 years (with Altera/Xilinx tools). It seems these tools are tolerant to a violation of pre-2008 standards, which have this, I believe, rather artificial constraint.
```
-- generic_test.vhdl

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
use std.textio.all;

entity GENERIC_WHEN is
        generic( FOO  :     std_logic_vector(1 downto 0) );
        port(    IN1  : in  std_logic_vector(1 downto 0);
                 OUT1 : out std_logic_vector(1 downto 0) );
end GENERIC_WHEN;

architecture BEHAVIOUR of GENERIC_WHEN is
begin
        PR1:
        process( IN1 )
                variable l : line;
        begin
                case IN1 is
                        when FOO    => OUT1 <= FOO; write(l, string'("FOO")); writeline(output, l);
                        --when "00"   => write(l, string'("00")); writeline(output, l);
                        when "01"   => write(l, string'("01")); writeline(output, l);
                        when others => write(l, string'("other")); writeline(output, l);
                end case;
        end process PR1;
end BEHAVIOUR;
```

```
-- generic_when_test1.vhdl

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

use std.textio.all;

--use work.GENERIC_WHEN.all;

entity TEST1 is
begin
end entity TEST1;

architecture BEHAVIOUR of TEST1 is
        component GENERIC_WHEN is
                generic( FOO  :     std_logic_vector(1 downto 0) );
                port(    IN1  : in  std_logic_vector(1 downto 0);
                         OUT1 : out std_logic_vector(1 downto 0) );
        end component GENERIC_WHEN;

        signal S1 : std_logic_vector(1 downto 0);
        signal S2 : std_logic_vector(1 downto 0);
begin

        GENERIC_WHEN_INST : GENERIC_WHEN
                generic map ( FOO  => "00" )
                port map (    IN1  => S1,
                              OUT1 => S2 );
        process
                variable l : line;
        begin
                S1 <= "01";
                writeline(output, l);
                wait;
        end process;

end architecture BEHAVIOUR;
```

```
-- generic_when_test2.vhdl

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

use std.textio.all;

--use work.GENERIC_WHEN.all;

entity TEST2 is
begin
end entity TEST2;

architecture BEHAVIOUR of TEST2 is
        component GENERIC_WHEN is
                generic( FOO  :     std_logic_vector(1 downto 0) );
                port(    IN1  : in  std_logic_vector(1 downto 0);
                         OUT1 : out std_logic_vector(1 downto 0) );
        end component GENERIC_WHEN;

        signal S1 : std_logic_vector(1 downto 0);
        signal S2 : std_logic_vector(1 downto 0);
begin

        GENERIC_WHEN_INST : GENERIC_WHEN
                generic map ( FOO  => "0" & "0")
                port map (    IN1  => S1,
                              OUT1 => S2 );
        process
                variable l : line;
        begin
                S1 <= "01";
                writeline(output, l);
                wait;
        end process;

end architecture BEHAVIOUR;
```
It fails:
```
~/progs/nvc/bin/nvc --work=work_nvc --std=2008 -a generic_when.vhdl generic_when_test1.vhdl generic_when_test2.vhdl
** Error: case choice must be locally static
	File generic_when.vhdl, Line 19
                            when FOO    => OUT1 <= FOO; write(l, string'("FOO")); w
                                 ^^^
make: *** [all] Error 1
```

I've commented out the relevant place in code:
```
diff -rupN nvc.orig/src/sem.c nvc/src/sem.c
--- nvc.orig/src/sem.c	2015-03-11 02:21:16.000000000 +0100
+++ nvc/src/sem.c	2015-03-13 17:04:52.127370398 +0100
@@ -118,6 +118,11 @@ static ident_t       unconstrained_i;
 static ident_t       protected_i;
 static bool          prefer_explicit = false;
 
+#define sem_ignore_error(t, ...) do {                 \
+      error_at(t ? tree_loc(t) : NULL , __VA_ARGS__); \
+      errors++;                                       \
+   } while (0)
+
 #define sem_error(t, ...) do {                        \
       error_at(t ? tree_loc(t) : NULL , __VA_ARGS__); \
       errors++;                                       \
@@ -5844,8 +5849,9 @@ static bool sem_check_case(tree_t t)
                if (!type_eq(tree_type(name), type))
                   sem_error(name, "case choice must have type %s",
                             sem_type_str(type));
-               else if (!sem_locally_static(name))
-                  sem_error(name, "case choice must be locally static");
+               //else if (!sem_locally_static(name))
+               //   sem_ignore_error(name, "case choice must be locally static");
+               //printf("(workaround)\n")
 
                tree_set_type(name, type);
             }
```
(By the way for some reason it's not enough to define sem_ignore_error() and use it in this way, but why?)

Then test1 pass. But test2 fails at the elaboration step:
```
~/progs/nvc/bin/nvc --work=work_nvc --std=2008 -L. -e --native test2
** Fatal: cannot handle tree kind T_CONCAT in rewrite_refs
	File generic_when_test2.vhdl, Line 25
make: *** [all] Error 1
```

Could you fix this for --std=2008 and maybe add something like --relax-std to available nvc options in order to allow such behavior also for pre-2008 standards?

Thanks

nickg, 2015-03-15:
I've implemented the VHDL-2008 rules allowing generics in locally static expressions and added a `--relax=generic-static` option that allows this with earlier standards.

I think your `sem_ignore_error` macro above would work if you remove the `errors++` bit.

closed 2015-03-15 by nickg

andreygursky, 2015-03-16:
Nick, thanks a lot!

> I think your sem_ignore_error macro above would work if you remove the errors++ bit.

Aha, I haven't noticed, that this counter is not only to track the count of errors, but is the actually reason to break with compiling.

But the second test generic_when_test2.vhdl
```vhdl
                generic map ( FOO  => "0" & "0")
```
still fails with:
```
~/progs/nvc/bin/nvc --work=work_nvc -L. -e --native test2
** Fatal: cannot handle tree kind T_CONCAT in rewrite_refs
	File generic_when_test2.vhdl, Line 25
make: *** [all] Error 1l
```

nickg, 2015-03-17:
Pushed a small change that makes this the same failure as issue #106