nickg/nvc #106: relax "case choice must be locally static" for sliced arrays

andreygursky, 2015-03-17:
Nick,

here is yet another candidate for relaxation:

```vhdl
-- sliced_array.vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
use std.textio.all;

package foo_const is
        constant foo1_c : integer := 0;
        constant foo2_c : integer := 7;
        constant addr1_c : std_logic_vector(foo2_c downto foo1_c) := x"02";
        constant addr2_c : std_logic_vector(foo2_c downto foo1_c) := x"03";
        constant addr3_c : std_logic_vector(foo2_c downto foo1_c) := x"04";
end foo_const;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
use std.textio.all;

use work.foo_const.all;

entity SLICED_ARRAY is
        port(    IN1  : in  std_logic_vector(7 downto 0);
                 OUT1 : out std_logic_vector(7 downto 0) );
end SLICED_ARRAY;

architecture BEHAVIOUR of SLICED_ARRAY is
begin
        PR1:
        process( IN1 )
                variable l : line;
        begin
                case IN1 is
                        -- when addr1_c => write(l, string'("OK")); writeline(output, l); -- OK
                        when addr2_c(7 downto 0) => write(l, string'("OK")); writeline(output, l); -- fails
                        when addr3_c(foo2_c downto foo1_c) => write(l, string'("OK")); writeline(output, l); -- fails
                        when others => write(l, string'("other")); writeline(output, l);
                end case;
        end process;
end BEHAVIOUR;
```

```vhdl
-- sliced_array_test.vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

use std.textio.all;

entity TEST is
begin
end entity TEST;

architecture BEHAVIOUR of TEST is
        component SLICED_ARRAY is
                port(    IN1  : in  std_logic_vector(7 downto 0);
                         OUT1 : out std_logic_vector(7 downto 0) );
        end component SLICED_ARRAY;

        signal S1 : std_logic_vector(7 downto 0);
        signal S2 : std_logic_vector(7 downto 0);
begin

        SLICED_ARRAY_INST : SLICED_ARRAY
                port map (    IN1  => S1,
                              OUT1 => S2 );
        process
                variable l : line;
        begin
                S1 <= x"10";
                writeline(output, l);
                wait;
        end process;

end architecture BEHAVIOUR;
```

After extending 
static bool sem_locally_static(tree_t t)
in src/sem.c with
```c
   if (kind == T_ARRAY_SLICE) {
      if (standard() >= STD_08 || relax & RELAX_LOCALLY_STATIC)
          return true;
   }
```
it fails at runtime in src/cgen.c at the assertion kind == T_REF, because kind is T_ARRAY_SLICE
```c
static void cgen_case_array(tree_t t, cgen_ctx_t *ctx)
...
            if ((kind != T_AGGREGATE) && (kind != T_LITERAL)) {
               assert(kind == T_REF);
               tree_t decl = tree_ref(name);
               assert(tree_kind(decl) == T_CONST_DECL);
               name = tree_value(decl);
               kind = tree_kind(name);
            }
```

By the way, it may be better to rename --relax option generic-static into locally-static and add this case to it, since there might be also other similar locally static cases.

Thanks,
Andrey

nickg, 2015-03-17:
Because fixing this requires changes to the code generator which I'm currently rewriting I'm going to leave this until after I merge the `new-codegen` branch. Hopefully will be in a few weeks.

andreygursky, 2015-03-28:
Good news.
By the way, do you plan to write a little bit about:
- approach you took for the first codegen (references?)
- what was not so good, resulted in a new-codegen
- are LLVM C bindings complete enough and maintained?

nickg, 2015-04-06:
I wrote a post on my blog about it: 
http://www.doof.me.uk/2015/04/06/new-code-generator/

andreygursky, 2016-07-15:
Hi Nick,
I'm revising now the status of this issue. It's still the same:
```
nvc: ../nvc.git/src/lower.c:3664: lower_case_array: Assertion `kind == T_REF' failed.
```
Could you sketch how this could be improved with a new code generator?

nickg, 2016-07-17:
I think I fixed it. Can you try with the latest version?

closed 2016-07-17 by nickg

andreygursky, 2016-07-19:
Yes, it works. Thanks! And there were indeed only little code changes necessary.

One thing about how you handle the lowering in the case statement. There is already a function for array slices ```lower_array_slice()```, but it doesn't get called in ```lower_case_...```, where the handling of ```T_ARRAY_SLICE``` looks much simpler.

In order to fix also #105 it seems an additional case for ```T_CONCAT``` should be added e.g. after the newly introduced ```T_ARRAY_SLICE``` in ```lower_case_find_choice_element```. Something like
```c
   case T_CONCAT:
      {
         const int nparams = tree_params(value);
         for (int i = 0; (i < nparams); i++) {
            // ?
         }
      }
      break;
```
I'm still missing more verbose docs about ```tree_t``` to fill this gap now. And ```lower_concat()``` is being called later anyway.

andreygursky, 2016-07-19:
P.S. test2 from 105.

nickg, 2016-07-21:
> One thing about how you handle the lowering in the case statement. There is already a function for array slices lower_array_slice(), but it doesn't get called in lower_case_..., where the handling of T_ARRAY_SLICE looks much simpler.

It's because the two functions are doing different things: `lower_array_slice` is generating code to calculate the value of an array slice expression. Whereas `lower_case_*` functions are building a state machine to decide which branch of the case statement to execute, when the expression is an array. The idea is than for N branches and M array elements, we generate M scalar case statements, which should be implemented much more efficiently in LLVM (using its `case` instruction) than N sequential `memcmp`s (N*M comparisons).

I added support for concatenation to this function. It was also fairly straightforward.