nickg/nvc #107: Reading procedure parameters of type inout fails

boehmerst, 2015-03-30:
Procedures with parameters of type inout cause an error:

  PROCEDURE pf(buf: INOUT LINE; s: INOUT REAL) IS
  BEGIN
    if buf/=NULL then if buf'LENGTH>0 then read(buf, s); end if; end if;
    DEALLOCATE(buf);
  END pf;

** Error: cannot read parameter S with mode OUT
	File stdio_h.vhd, Line 1137
        if buf/=NULL then if buf'LENGTH>0 then read(buf, s); end if; end if;
                                                                                        ^

boehmerst, 2015-03-30:
Maybe this relates to the signature of the procedure read(buf, s) beeing called since nvc does not complain about parameter buf but about parameter s. Both are of type inout in the caller not in the callee.

procedure read (l : inout line; value : out real );

nickg, 2015-03-30:
Should be fixed by the above commit

closed 2015-03-30 by nickg