nickg/nvc #108: make check shows strange failure

postoroniy, 2015-03-31:
Hi Nick,

I built from last master, ran check and got this
 make check
  CC       test/test_lib.o
  CCLD     bin/test_lib
  CC       test/test_ident.o
  CCLD     bin/test_ident
  CC       test/test_parse.o
  CCLD     bin/test_parse
  CC       test/test_sem.o
  CCLD     bin/test_sem
  CC       test/test_simp.o
  CCLD     bin/test_simp
  CC       test/test_elab.o
  CCLD     bin/test_elab
  CC       test/test_heap.o
  CCLD     bin/test_heap
  CC       test/test_hash.o
  CCLD     bin/test_hash
  CC       test/test_group.o
  CCLD     bin/test_group
  CC       test/test_bounds.o
  CCLD     bin/test_bounds
  CC       test/test_value.o
  CCLD     bin/test_value
  CC       test/test_lower.o
  CCLD     bin/test_lower
Running suite(s): lib
100%: Checks: 3, Failures: 0, Errors: 0
PASS: bin/test_lib
Running suite(s): ident
100%: Checks: 15, Failures: 0, Errors: 0
PASS: bin/test_ident
Running suite(s): parse
100%: Checks: 31, Failures: 0, Errors: 0
PASS: bin/test_parse
Running suite(s): sem
100%: Checks: 38, Failures: 0, Errors: 0
PASS: bin/test_sem
Running suite(s): simplify
100%: Checks: 5, Failures: 0, Errors: 0
PASS: bin/test_simp
Running suite(s): elab
100%: Checks: 15, Failures: 0, Errors: 0
PASS: bin/test_elab
Running suite(s): heap
100%: Checks: 3, Failures: 0, Errors: 0
PASS: bin/test_heap
Running suite(s): hash
100%: Checks: 3, Failures: 0, Errors: 0
PASS: bin/test_hash
Running suite(s): group
100%: Checks: 10, Failures: 0, Errors: 0
PASS: bin/test_group
Running suite(s): bounds
100%: Checks: 4, Failures: 0, Errors: 0
PASS: bin/test_bounds
Running suite(s): value
100%: Checks: 2, Failures: 0, Errors: 0
PASS: bin/test_value
Running suite(s): lower

Name       :memset:bar$uINT_VECTOR;P
Kind       function
Context    WORK.MEMSET.elab
Blocks     4
Registers  27
Types      12
Variables  1
  V                                     // [*] : -2^31..2^31-1 
Result     [*] : -2^31..2^31-1
Parameters 1
  r0    X                               // -2^31..2^31-1 => 1..2^31-1 
Begin
   0: // Begin dynamic aggregrate line 15 
      r1 := const 2147483647            // -2^31..2^31-1 => 2^31-1 
      // Dead const definition of r2 
      r3 := const 1                     // -2^31..2^31-1 => 1 
      r4 := sub r0 - r3                 // -2^31..2^31-1 => 0..2147483646 
      r5 := add r4 + r3                 // -2^31..2^31-1 => 1..2^31-1 
      r6 := cast r5                     // # => 1..2^31-1 
      r7 := const 0                     // # => 0 
      r8 := cmp r6 < r7
      r9 := select r8 then r7 else r6   // # 
      r10 := alloca heap r9             // @<-2^31..2^31-1> => -2^31..2^31-1 
      r11 := alloca                     // @<#> => # 
      r11 := store indirect r7
      jump 1
   1: r12 := load indirect r11          // # 
      r13 := cmp r12 >= r9
      cond r13 then 3 else 2
   2: r14 := add r10 + r12              // @<-2^31..2^31-1> => -2^31..2^31-1 
      r14 := store indirect r1
      r15 := const 1                    // # => 1 
      r16 := add r12 + r15              // # => -2147483647..2147483648 
      r11 := store indirect r16
      jump 1
   3: // End dynamic aggregrate line 15 
      // Dead wrap definition of r17 
      r18 := const 1                    // -2^31..2^31-1 => 1 
      // Elided index check for r18 
      // Elided index check for r0 
      r19 := sub r0 - r18               // -2^31..2^31-1 => 0..2147483646 
      r20 := add r19 + r18              // -2^31..2^31-1 => 1..2^31-1 
      r21 := cast r20                   // # => 1..2^31-1 
      r22 := const 0                    // # => 0 
      r23 := cmp r21 < r22
      r24 := select r23 then r22 else r21 // # 
      r25 := const 0                    // 0..1 => 0 
      r26 := wrap r10 [r18 r0 r25]      // [*] : -2^31..2^31-1 
      V := store r26
      return r26

96%: Checks: 31, Failures: 1, Errors: 0
../test/test_lower.c:79:F:Core:test_memset:0: expected op 1 in block 0 to have constant 0 but has 0
FAIL: bin/test_lower
than shows that all tests were passed and at the end this:
==============================================
1 of 13 tests failed
Please report to Nick Gasson <nick@nick.me.uk>
==============================================
make[1]: *** [check-TESTS] Error 1
make: *** [check-am] Error 2

postoroniy, 2015-03-31:
some extra
llc --version
LLVM (http://llvm.org/):
  LLVM version 3.5.0
  Optimized build with assertions.
  Built Sep 14 2014 (12:18:00).
  Default target: i386-pc-linux-gnu
  Host CPU: core2

nickg, 2015-03-31:
That was weird: it's actually a bug in how it parses based literals that behaves differently on 32- vs. 64-bit machines.

closed 2015-03-31 by nickg