nickg/nvc #117: Error while running simulations (archlinux, llvm 3.6)

jck, 2015-04-11:
LLVM ERROR: Target does not support MC emission!

Analysis and elaboration seem to work.

jck, 2015-04-11:
Output of make check:
```
  CCLD     bin/test_parse
/usr/bin/ld: lib/libnvc.a(lexer.o): undefined reference to symbol 'pow@@GLIBC_2.2.5'
/usr/lib/libm.so.6: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
Makefile:1172: recipe for target 'bin/test_parse' failed
make[1]: *** [bin/test_parse] Error 1
Makefile:2076: recipe for target 'check-am' failed
make: *** [check-am] Error 2
```

jck, 2015-04-11:
Using the '--native' flag during elaboration seems to work around the issue.

nickg, 2015-04-11:
Yes I think the LLVM JIT interface changed between 3.5 and 3.6. I'll install 3.6 and fix it.

nickg, 2015-04-11:
Can you see if the above commit fixes the `make check` link error?

nickg, 2015-04-11:
JIT works with LLVM 3.6 now.

closed 2015-04-11 by nickg

jck, 2015-04-12:
Thanks, I just tried nvc with MyHDL's testsuite and all tests passed!

nickg, 2015-04-12:
Great, thanks for testing that.