nickg/nvc #118: autogen.sh doesn't work on debian stable (automake 1.11)

fmhess, 2015-04-11:
The following patch makes autogen.sh work on debian stable by removing the "--serial-tests" option from automake invocation when using automake 1.11.

```
$ git diff
diff --git a/autogen.sh b/autogen.sh
index 2caa76a..b3b9db8 100755
--- a/autogen.sh
+++ b/autogen.sh
@@ -1,3 +1,10 @@
 #!/bin/sh
 cd $(git rev-parse --show-toplevel)
+am_ver=`automake --version | sed 1q`
+  # Autmake 1.11.x doesn't grasp the 'serial-tests' option.
+  case $am_ver in
+    *\ 1.11.*) echo 'm4_define([SERIAL_TESTS_OPTION], [])';;
+    *) echo 'm4_define([SERIAL_TESTS_OPTION], [serial-tests])';;
+  esac > m4/serial_tests_option.m4
+  
 autoreconf --force --install -I m4
diff --git a/configure.ac b/configure.ac
index 7423628..07e9c8a 100644
--- a/configure.ac
+++ b/configure.ac
@@ -1,8 +1,9 @@
 AC_INIT([nvc], [0.1],
         [Nick Gasson <nick@nick.me.uk>],
         [nvc])
-AM_INIT_AUTOMAKE([1.12 -Wall -Wno-extra-portability color-tests
-                  serial-tests foreign subdir-objects])
+m4_include([m4/serial_tests_option.m4])
+AM_INIT_AUTOMAKE([1.11 -Wall -Wno-extra-portability color-tests
+                  foreign subdir-objects] SERIAL_TESTS_OPTION)
 AM_SILENT_RULES([yes])
 
 AC_CONFIG_MACRO_DIR([m4])
```

nickg, 2015-04-11:
Is it possible to do this without putting the logic in the `autogen.sh` script? Problem is that it will break if people do a `git pull` and don't re-run the script. Does something like this work?
https://www.redhat.com/archives/libguestfs/2013-February/msg00102.html

nickg, 2015-04-11:
I tried the workaround from the link above which is entirely within `configure.ac`. Can you try it out?

fmhess, 2015-04-11:
Yes, your workaround works fine for me.

closed 2015-04-11 by nickg