nickg/nvc #121: cgen_procedure assertion, forward reference procedure call.

ikwzm, 2015-04-13:
````vhdl
-- a_ng.vhd
library ieee;
use     ieee.std_logic_1164.all;
package A is
    procedure PROC_A(I:in integer; O:out integer; Z:out boolean);
    procedure PROC_B(I:in integer; O:out integer; Z:out boolean);
    procedure PROC_C(I:in integer; O:out integer; Z:out boolean);
end package;
library ieee;
use     ieee.std_logic_1164.all;
package body A is
    procedure PROC_A(I:in integer; O:out integer; Z:out boolean) is
    begin
        PROC_B(I,O,Z);
    end procedure;
    procedure PROC_B(I:in integer; O:out integer; Z:out boolean) is
    begin
        PROC_C(I,O,Z);
    end procedure;
    procedure PROC_C(I:in integer; O:out integer; Z:out boolean) is
    begin 
        O := I;
        Z := (I = 0);
    end procedure;
end package body;
````

````
ikwzm@-vm-ubuntu:~/work/nvc/nvc-test$ ../build/bin/nvc -L ../build/lib -a a_ng.vhd
nvc: ../src/cgen.c:2711: cgen_procedure: Assertion `LLVMGetNamedFunction(module, istr(vcode_unit_name())) == ((void *)0)' failed.

*** Caught signal 6 (SIGABRT) ***

-------- STACK TRACE --------
/lib/x86_64-linux-gnu/libc.so.6(gsignal+0x39) [0x7ff26534abb9]
/lib/x86_64-linux-gnu/libc.so.6(gsignal+0x39) [0x7ff26534abb9]
/lib/x86_64-linux-gnu/libc.so.6(abort+0x148) [0x7ff26534dfc8]
/lib/x86_64-linux-gnu/libc.so.6(+0x2fa76) [0x7ff265343a76]
/lib/x86_64-linux-gnu/libc.so.6(+0x2fb22) [0x7ff265343b22]
../build/bin/nvc() [0x47133a]
../build/bin/nvc(cgen+0x5f5) [0x471a15]
../build/bin/nvc(main+0x1263) [0x411ac3]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf5) [0x7ff265335ec5]
../build/bin/nvc() [0x411d75]
````

ikwzm, 2015-04-13:
assertion occrued at cgen_procedure()...

````c
static void cgen_procedure(LLVMTypeRef display_type)
{
   assert(vcode_unit_kind() == VCODE_UNIT_PROCEDURE);
   assert(LLVMGetNamedFunction(module, istr(vcode_unit_name())) == NULL);

   LLVMValueRef fn = LLVMAddFunction(module, istr(vcode_unit_name()),
                                     cgen_subprogram_type(display_type, true));
````
When cgen_procedure() was changed this, it didn't occur any more.

````c
static void cgen_procedure(LLVMTypeRef display_type)
{
   assert(vcode_unit_kind() == VCODE_UNIT_PROCEDURE);
   LLVMValueRef fn = LLVMGetNamedFunction(module, istr(vcode_unit_name()));

   if (fn == NULL)
       fn = LLVMAddFunction(module, istr(vcode_unit_name()),
                                     cgen_subprogram_type(display_type, true));

````

My change does not have the correct whether confidence....

nickg, 2015-04-13:
Yes that's the correct change to make it match `cgen_function`. Thanks.

closed 2015-04-13 by nickg