nickg/nvc #123: nvc: src/vcode.c:2632: emit_extern_var: Assertion `active_unit->kind == VCODE_UNIT_CONTEXT' failed.

kraigher, 2015-04-13:
```vhdl
package nested_function_bug is
  procedure proc(param : integer);
end package;

package body nested_function_bug is
  procedure proc(param : integer) is
    variable foo : bit_vector(0 to param);
    function nested_function return integer is -- Also fails with "impure function"
    begin
      return foo'length;
    end;
  begin
  end;
end package body;
```

```shell
> nvc -a function.vhd
nvc: src/vcode.c:2632: emit_extern_var: Assertion `active_unit->kind == VCODE_UNIT_CONTEXT' failed.

*** Caught signal 6 (SIGABRT) ***

-------- STACK TRACE --------
/lib/x86_64-linux-gnu/libc.so.6(gsignal+0x39) [0x7f6140cbccc9]
/lib/x86_64-linux-gnu/libc.so.6(gsignal+0x39) [0x7f6140cbccc9]
/lib/x86_64-linux-gnu/libc.so.6(abort+0x148) [0x7f6140cc00d8]
/lib/x86_64-linux-gnu/libc.so.6(+0x2fb86) [0x7f6140cb5b86]
/lib/x86_64-linux-gnu/libc.so.6(+0x2fc32) [0x7f6140cb5c32]
nvc(emit_extern_var+0x8e) [0x46552e]
nvc() [0x456d57]
nvc() [0x456d90]
nvc() [0x45823f]
nvc() [0x45aa7e]
nvc() [0x45b1f5]
nvc() [0x4578aa]
nvc() [0x458fde]
nvc() [0x45e18d]
nvc() [0x45ed09]
-----------------------------

```

Also other than the crash a semantic error message would be useful in this case:
```vhdl
package nested_function_bug is
  procedure proc(param : integer);
end package;

package body nested_function_bug is
  procedure proc(param : integer) is
    variable foo : integer := param;
    function nested_function return integer is
    begin
      return foo;
    end;
  begin
  end;
end package body;
```

instead of:

```shell
nvc -a function.vhd
nvc: src/vcode.c:2632: emit_extern_var: Assertion `active_unit->kind == VCODE_UNIT_CONTEXT' failed.

*** Caught signal 6 (SIGABRT) ***

-------- STACK TRACE --------
/lib/x86_64-linux-gnu/libc.so.6(gsignal+0x39) [0x7f795cf4ccc9]
/lib/x86_64-linux-gnu/libc.so.6(gsignal+0x39) [0x7f795cf4ccc9]
/lib/x86_64-linux-gnu/libc.so.6(abort+0x148) [0x7f795cf500d8]
/lib/x86_64-linux-gnu/libc.so.6(+0x2fb86) [0x7f795cf45b86]
/lib/x86_64-linux-gnu/libc.so.6(+0x2fc32) [0x7f795cf45c32]
nvc(emit_extern_var+0x8e) [0x46552e]
nvc() [0x456d57]
nvc() [0x456d90]
nvc() [0x45823f]
nvc() [0x458fde]
nvc() [0x45e18d]
nvc() [0x45ed09]
nvc() [0x45ee3b]
nvc(lower_unit+0x256) [0x45f136]
nvc(main+0x1255) [0x411ab5]
-----------------------------
```

other simulators say:
```shell
** Cannot reference variable "foo" inside pure function "nested_function".
```

kraigher, 2015-04-14:
To clarify: The first case is legal and the second case is illegal and should give a semantic error.

nickg, 2015-04-14:
Both of these should be fixed now. Now with the pure version I get:

```
** Error: invalid reference to FOO inside pure function NESTED_FUNCTION
        File ../test/regress/issue123.vhd, Line 10
          return foo'length;
                 ^^^
```

closed 2015-04-14 by nickg

kraigher, 2015-04-15:
@nickg foo'length should actually work since it is equivalent with a constant in the outer scope of the pure function. It works in other simulators at least. It is only the second vhdl code I posted that should trigger a semantic error. The first code should work.

nickg, 2015-04-16:
It's not immediately obvious to me that that's the case: in section 4.3 of the 2008 LRM it says:

> [...] if a pure function subprogram contains a reference to an explicitly declared signal or variable
object, or a slice or subelement (or slice thereof) of an explicit signal, then that object shall be declared within the declarative region formed by the function; [...]

And the glossary defines "a reference" as:

> Access to a named entity. Every appearance of a designator (a name, character literal, or operator
symbol) is a reference to the named entity denoted by the designator, unless the designator appears in a library clause or use clause. 

So from my reading `foo'length` contains a reference to `foo` which is a variable not declared in the function's declarative region.

kraigher, 2015-04-17:
@nickg I agree with you that the LRM forbids it. I checked in both Vivado and Sigasi and they produce an error. Only ModelSim accepts the code. In any case it is easy to rewrite the violating code by creating an explicit constant in the outer scope assigning it the foo'length attribute.