nickg/nvc #126: simulation produces wrong output

fmhess, 2015-04-14:
The following test program fails.  What should happen is clk_count counts down from 6.  Instead, the first time it is supposed to decrement it becomes undefined.

```
entity nvc_bug3 is
end nvc_bug3;

architecture structural of nvc_bug3 is

	signal clk : std_logic;
	signal clk_count : std_logic_vector (2 downto 0);
	signal reset : std_logic;
	
	shared variable test_finished : boolean := false;

	constant sysclk_half_period : time := 5 ns;

begin

	process
	begin
		if(test_finished) then
			wait;
		end if;
		
		clk <= '0';
		wait for sysclk_half_period;
		clk <= '1';
		wait for sysclk_half_period;
	end process;

	-- Process to divide system clock by six.
	process(clk) is
	begin
		if reset = '0' then
			clk_count <= "110";
		elsif clk'event and clk = '1' then
			if clk_count = "001" then
				clk_count <= "110";
			else
				clk_count <= clk_count - "001";
			end if;
		end if;
	end process;
  
  process
  begin
		reset <= '0';
		wait for 20 ns;
		reset <= '1';
		
		wait for 100 ns;
		for i in 0 to 2 loop
			assert clk_count(i) /= 'U' report "clk_count is undefined!" severity failure;
		end loop;
		
		test_finished := true;
		assert false report "test finished" severity note;
		wait;
	end process;
end architecture structural;

```

fmhess, 2015-04-15:
Here is a smaller testcase.  The issue is subtracting an unsigned from a std_logic_vector doesn't work.  There is a similar bug with the addition operator.

```
library ieee;
use ieee.std_logic_1164.all;
use ieee.std_logic_unsigned."-";

entity nvc_bug3 is
end nvc_bug3;

architecture structural of nvc_bug3 is

	signal clk_count : std_logic_vector (2 downto 0);
begin

  process
  begin
		clk_count <= "110";
		wait for 1 us;
		clk_count <= clk_count - 1;
		wait for 1 us;
		assert clk_count = "101" severity error;

		wait;
	end process;
end architecture structural;

```

fmhess, 2015-04-15:
If I add 
```
use ieee.std_logic_arith.all;
```
to the testcase, it passes.

fmhess, 2015-04-15:
Or, more specifically I can add
```
use ieee.std_logic_arith.unsigned;
```
to the testcase to make it pass.  The definition of the "-" operator in std_logic_unsigned.vhd uses the unsigned type defined in std_logic_arith.  std_logic_unsigned.vhd has a "use IEEE.std_logic_arith.all;" but for whatever reason that is not enough, and I have to have the use clause directly in my code as well.

postoroniy, 2015-04-16:
I had the same issue, but I  decided to use 
1) numeric_std library
2) signed and unsigned types of signals for counters/addresses and etc. which is not problem at all
3) when I need std_logic_vector I use conversion from signed/unsigned into std_logic_vector. which is fine as well.
Also, I remember that std_logic_arith is something old not recommended to use , and I am using as I mentioned early numeric_std.
Anyway, there's some issue in nvc to work std_logic_arith correctly.

nickg, 2015-04-16:
Thanks for narrowing it down. I've checked your original program works now.

Please report any issues you run into using the Synopsys packages - as @postoroniy says they're not often used in new designs so don't get tested as much as `numeric_std`, etc.

closed 2015-04-16 by nickg