nickg/nvc #128: sem_hoist_for_loop_var() assertion, case statement with alternative includes some choises.

ikwzm, 2015-04-18:
````vhdl
-- a_ng.vhd
package A_NG is
    procedure PROC(SEL: in integer);
end package;
package body A_NG is
    procedure PROC(SEL: in integer) is
    begin
        case SEL is
            when 0 | 1 => 
                for i in 0 to 3 loop
                end loop;
            when others =>
                null;
        end case;
    end procedure;
end package body;
````

````
ikwzm@vm-ubuntu:~/work/nvc/issue-4-test$ ../build/bin/nvc -L ../build/lib -a a_ng.vhd
nvc: ../src/sem.c:2354: sem_hoist_for_loop_var: Assertion `ndecls == 1' failed.

*** Caught signal 6 (SIGABRT) ***

-------- STACK TRACE --------
/lib/x86_64-linux-gnu/libc.so.6(gsignal+0x39) [0x7fe313bc3bb9]
/lib/x86_64-linux-gnu/libc.so.6(gsignal+0x39) [0x7fe313bc3bb9]
/lib/x86_64-linux-gnu/libc.so.6(abort+0x148) [0x7fe313bc6fc8]
/lib/x86_64-linux-gnu/libc.so.6(+0x2fa76) [0x7fe313bbca76]
/lib/x86_64-linux-gnu/libc.so.6(+0x2fb22) [0x7fe313bbcb22]
../build/bin/nvc() [0x434ff8]
../build/bin/nvc(object_visit+0x298) [0x452fd8]
../build/bin/nvc(object_visit+0xff) [0x452e3f]
../build/bin/nvc(object_visit+0x1fc) [0x452f3c]
../build/bin/nvc(object_visit+0xff) [0x452e3f]
../build/bin/nvc(object_visit+0xff) [0x452e3f]
../build/bin/nvc(tree_visit_only+0x3b) [0x42ea1b]
../build/bin/nvc(sem_check+0x10db) [0x436f1b]
../build/bin/nvc() [0x43be5e]
../build/bin/nvc(sem_check+0xe69) [0x436ca9]
-----------------------------
````

ikwzm, 2015-04-18:
sem_check_case() is calling sem_check().

````C
static bool sem_check_case(tree_t t)
{
    ...
   for (int i = 0; i < nassocs; i++) {
      tree_t a = tree_assoc(t, i);
      switch (tree_subkind(a)) {
      case A_OTHERS:
          ...
      case A_NAMED:
          ...
      case A_RANGE:
          ...
      default:
          ...
      }
      ok = sem_check(tree_value(a)) && ok;
   }
   return ok;
}
````

For example "case 1 | 2 =>"
"sem_check(tree_value(a))" is evaluated two times of same tree.
As a result, in "sem_hoist_for_loop_var(tree_t t, void *context)", "tree_decls(t)" is 2.

nickg, 2015-04-19:
I've been meaning to change the way for loops are handled for a while now so this seems like a good reason. That function's been removed so won't assert any more.

closed 2015-04-19 by nickg