nickg/nvc #129: Cygwin library build issue

michg, 2015-04-21:
After commit "Move context walking logic from link to elaboration stage. Issue #126 " the building of
the libraries fails on Cygwin.
[image]

michg, 2015-04-21:
It seems to be an make "order" issue. After repeating "make" several times installation is possible.

nickg, 2015-04-21:
Was broken on Cygwin due to its requirement to link an extra static library. Builds OK now on a Windows 8 PC.

closed 2015-04-21 by nickg

michg, 2015-04-22:
Thanks, works fine now.

michg, 2015-04-24:
Hi Nick,
I have two more open issues, which I am not able to localize in my source code. One happens at the -r stage and throws an exception: 0 [main] nvc 3716 cygwin_exception::open_stackdumpfile: dumping stack trace to nvc.exe.stackdump. The other one is at elaboration: assertion "!tree_has code(body)" failed : file "../src/lower.c", line 4127, function: lower_func_body. Could you give me some hints for finding the relevant source code?. Thanks a lot.

nickg, 2015-04-25:
Does the stack dump file `nvc.exe.stackdump` exist? The automatic backtrace code has never worked very well on Cygwin. Otherwise, is it possible to run it in GDB?

michg, 2015-04-27:
Yes, the stackdump exists:
Exception: STATUS_ACCESS_VIOLATION at eip=6C381130
eax=00000226 ebx=8159F3C8 ecx=9C19FAD0 edx=00000000 esi=0028CAC0 edi=00000087
ebp=0028CB58 esp=0028CAC0 program=d:\nvc27415bin\bin\nvc.exe, pid 6688, thread main
cs=0023 ds=002B es=002B fs=0053 gs=002B ss=002B
Stack trace:
Frame     Function  Args
0028CB58  6C381130 (8159F3C8, 00000000, 8159F398, 8157A8A8)
End of stack trace
 I also executed the code with gdb:
[image]

nickg, 2015-04-27:
It's crashing while it's resetting the design (i.e. assigning initial values to signals, etc.) which is hopefully easier to track down than a crash while running the simulation. One way to track down the cause, if your design is quite hierarchical, would be to remove instantiations until it no longer crashes.

Does "make check" run without errors on your system? I've seen similar crashes on Cygwin before when the `IEEE.MATH_REAL` and `IEEE.MATH_COMPLEX` libraries are used (but these should be fixed now).

michg, 2015-04-28:
The "make check" is errorfree. As the design is quite large it may take a bit to find the problem..