nickg/nvc #130: Can't found the procedure name by recursive call and forward reference call.

ikwzm, 2015-04-22:
````VHDL
-- a_ng.vhd
package A_NG is
end package;
package body A_NG is
    procedure PROC(S: in integer; Q: out integer) is
    begin
        PROC(S,Q);
    end procedure;
end package body;
````

````VHDL
-- a_ok.vhd
package A_OK is
end package;
package body A_OK is
    procedure PROC(S: in integer; Q: out integer);
    procedure PROC(S: in integer; Q: out integer) is
    begin
        PROC(S,Q);
    end procedure;
end package body;
````

````
ikwzm@vm-ubuntu:~/work/nvc/issue-6-test$ ../build/bin/nvc -L ../build/lib -a a_ng.vhd
** Error: undefined procedure PROC
	File a_ng.vhd, Line 6
            PROC(S,Q);
            ^^^^^^^^^^

````

ikwzm, 2015-04-22:
error occured at sem_check_proc_body()

````C
static bool sem_check_proc_body(tree_t t)
{
   if (!sem_check_proc_ports(t))
      return false;

   scope_apply_prefix(t);

   // If there is no declaration for this procedure add to the scope
   if (!sem_check_duplicate(t, T_PROC_DECL)) {
      if (!scope_insert(t))
         return false;
   }

   scope_push(NULL);
   top_scope->subprog = t;

````

When sem_check_proc_body() was changed this, it didn't occur any more.

````C
static bool sem_check_proc_body(tree_t t)
{
   if (!sem_check_proc_ports(t))
      return false;

   ident_t unqual = top_scope->prefix ? tree_ident(t) : NULL;
   scope_apply_prefix(t);

   // If there is no declaration for this procedure add to the scope
   if (!sem_check_duplicate(t, T_PROC_DECL)) {
      if (!scope_insert(t))
         return false;
   }

   scope_push(NULL);
   top_scope->subprog = t;

   if (unqual != NULL)
      scope_insert_alias(t, unqual);

````

It was changed like sem_check_func_body().

````C
static bool sem_check_func_body(tree_t t)
{
   if (!sem_check_func_ports(t))
      return false;

   ident_t unqual = top_scope->prefix ? tree_ident(t) : NULL;
   scope_apply_prefix(t);

   // If there is no declaration for this function add to the scope
   if (!sem_check_duplicate(t, T_FUNC_DECL)) {
      if (!scope_insert(t))
         return false;
   }

   scope_push(NULL);
   top_scope->subprog = t;

   if (unqual != NULL)
      scope_insert_alias(t, unqual);

````

nickg, 2015-04-22:
Yes that looks like the right fix. I've committed it.

closed 2015-04-22 by nickg