nickg/nvc #131: When the type of the name which overlapped a subprogram is declared, the wrong type is referred to.

ikwzm, 2015-04-23:
````VHDL
-- a_ng.vhd
package A_NG is
    type STATE_TYPE is (STATE_IDLE, STATE_0, STATE_1);
end package;
package body A_NG is
    procedure PROC is
        type     STATE_TYPE is (STATE_IDLE, STATE_A, STATE_B);
        variable state       :  STATE_TYPE;
    begin
        state := STATE_A;
    end procedure;
end package body;
````

````VHDL
-- a_ok.vhd
package A_OK is
    type STATE_TYPE is (STATE_IDLE, STATE_0, STATE_1);
end package;
package body A_OK is
    procedure PROC is
        type     LOCAL_STATE_TYPE is (STATE_IDLE, STATE_A, STATE_B);
        variable state :  LOCAL_STATE_TYPE;
    begin
        state := STATE_A;
    end procedure;
end package body;

````

````
ikwzm@vm-ubuntu:~/work/nvc/issue-7-test$ ../build/bin/nvc -L ../build/lib -a a_ng.vhd
** Error: name STATE_A cannot be used in this context
              STATE_TYPE
	File a_ng.vhd, Line 9
            state := STATE_A;
                     ^^^^^^^
````

ikwzm, 2015-04-23:
````C
static bool sem_declare(tree_t decl, bool add_predefined)
{
   // Handle special cases of scope insertion such as enumeration
   // literals, physical unit names, and predefined types

   // Certain kinds of declarations like components do not have
   // a type
   if (tree_kind(decl) == T_COMPONENT)
      return scope_insert(decl);

   // Resolve the base type if necessary
   if (!sem_check_subtype(decl, tree_type(decl), NULL))
      return false;

   // If this is a full type declarataion then replace any previous
   // incomplete type declaration
   tree_t forward = scope_find(tree_ident(decl));
   if (forward != NULL && tree_kind(forward) == T_TYPE_DECL) {
      type_t incomplete = tree_type(forward);
      if (type_kind(incomplete) == T_INCOMPLETE) {
         // Replace the incomplete type with the one we are defining
         type_replace(incomplete, tree_type(decl));
         tree_set_type(decl, incomplete);

         // Create a new incomplete type and attach that to the
         // forward declaration: this is useful when we serialise
         // the tree to avoid circular references
         type_t ni = type_new(T_INCOMPLETE);
         type_set_ident(ni, type_ident(incomplete));
         tree_set_type(forward, ni);

         scope_replace(forward, decl);
      }
   }
   else if (!scope_insert(decl))
      return false;

````

If forward != NULL and tree_kind(forward) == T_TYPE_DECL and type_kind(tree_type(forward)) != T_INCOMPLETE,  isn't it necessary to do "scope_insert(decl)" ?

nickg, 2015-04-25:
Yes that's right. Thanks for tracking it down.

closed 2015-04-25 by nickg