nickg/nvc #132: When type of generic is record type,  can not found record member of generic in architecture.

ikwzm, 2015-04-23:
````VHDL
-- a_pkg.vhd
package A_PKG is
    type POS_TYPE is record
         X : integer;
         Y : integer;
    end record;
end package;
````

````VHDL
-- a_ng.vhd
use WORK.A_PKG.all;
entity A_NG is
    generic (POS: POS_TYPE := (X => 0, Y => 0));
end entity;
architecture MODEL of A_NG is
    procedure GET_X(X:out integer) is
    begin
        X := POS.X;
    end procedure;
begin
end architecture;
````

````VHDL
-- a_ok.vhd
use WORK.A_PKG.all;
entity A_OK is
end entity;
architecture MODEL of A_OK is
    constant  POS : POS_TYPE := (X => 0, Y => 0);
    procedure GET_X(X:out integer) is
    begin
        X := POS.X;
    end procedure;
begin
end architecture;
````

````
ikwzm@vm-ubuntu:~/work/nvc/issue-8-test$ ../build/bin/nvc -L ../build/lib -a a_pkg.vhd
ikwzm@vm-ubuntu:~/work/nvc/issue-8-test$ ../build/bin/nvc -L ../build/lib -a a_ok.vhd
ikwzm@vm-ubuntu:~/work/nvc/issue-8-test$ ../build/bin/nvc -L ../build/lib -a a_ng.vhd
** Error: no visible declaration for POS.X
	File a_ng.vhd, Line 8
            X := POS.X;
                 ^^^
````

ikwzm, 2015-04-23:
````C
static bool sem_check_arch(tree_t t)
{
    ...
   const int nports = tree_ports(e);
   for (int n = 0; n < nports; n++) {
      tree_t p = tree_port(e, n);
      scope_insert(p);

      type_t type = tree_type(p);
      if (type_is_record(type))
         sem_declare_fields(type, tree_ident(p));
   }

   const int ngenerics = tree_generics(e);
   for (int n = 0; n < ngenerics; n++)
      scope_insert(tree_generic(e, n));

    ...
````

When is it also necessary to do a loop of generics like a loop of ports?

````C
   const int ngenerics = tree_generics(e);
   for (int n = 0; n < ngenerics; n++) {
      tree_t g = tree_generic(e, n);
      scope_insert(g);

      type_t type = tree_type(g);
      if (type_is_record(type))
         sem_declare_fields(type, tree_ident(g));
   }

````

ikwzm, 2015-04-23:
Besides similar correction may also be needed...

````C
static bool sem_check_proc_body(tree_t t)
{
...
   const int nports = tree_ports(t);
   for (int i = 0; i < nports; i++)
      ok = scope_insert(tree_port(t, i)) && ok;
...
}
````

````C
static bool sem_check_generics(tree_t t)
{
...
   if (ok) {
      // Make generics visible in this region
      for (int n = 0; n < ngenerics; n++)
         ok = scope_insert(tree_generic(t, n)) && ok;
   }
...
}

nickg, 2015-04-23:
Yes the call to `sem_declare_fields` was missing in some places. I've refactored the code a bit to avoid this - it now just calls a single `sem_make_visible` function in these places.

closed 2015-04-23 by nickg