nickg/nvc #133: When the constant of record type was defined by a package,  can not found record member of the constant in architecture.

ikwzm, 2015-04-24:
````VHDL
-- a_pkg.vhd
package A_PKG is
    type POS_TYPE is record
         X : integer;
         Y : integer;
    end record;
    constant  POS_ZERO : POS_TYPE := (X => 0, Y=> 0);
end package;
````

````VHDL
use WORK.A_PKG.all;
entity A_NG is
end entity;
architecture MODEL of A_NG is
    procedure GET_ZERO_X(X:out integer) is
    begin
        X := POS_ZERO.X;
    end procedure;
begin
end architecture;
````

````
ikwzm@vm-ubuntu:~/work/nvc/issue-9-test$ ../build/bin/nvc -L ../build/lib -a a_pkg.vhd
ikwzm@vm-ubuntu:~/work/nvc/issue-9-test$ ../build/bin/nvc -L ../build/lib -a a_ng.vhd
** Error: record type POS_TYPE has no field X
	File a_ng.vhd, Line 7
            X := POS_ZERO.X;
                 ^^^^^^^^

````

ikwzm, 2015-04-24:
````C
static bool scope_import_unit(ident_t unit_name, lib_t lib,
                              bool all, const loc_t *loc)
{
...
      if (!unqual_only) {
         if (!sem_declare(decl, true))
            return false;

         ident_t pqual = ident_from(dname, '.');
         if (pqual != NULL)
            scope_insert_alias(decl, pqual);
      }

      if (all) {
         ident_t unqual = ident_rfrom(dname, '.');
         if (unqual != NULL)
            scope_insert_alias(decl, unqual);
      }
   }
...
}
````

I corrected as issue #132.

````C
static bool scope_import_unit(ident_t unit_name, lib_t lib,
                              bool all, const loc_t *loc)
{
...
      if (!unqual_only) {
         if (!sem_declare(decl, true))
            return false;

         ident_t pqual = ident_from(dname, '.');
         if (pqual != NULL) {
            scope_insert_alias(decl, pqual);

            if ((kind == T_CONST_DECL) || (kind == T_SIGNAL_DECL) || (kind == T_VAR_DECL)) 
               if (type_is_record(tree_type(decl)))
                  sem_declare_fields(tree_type(decl), pqual);
         }
      }

      if (all) {
         ident_t unqual = ident_rfrom(dname, '.');
         if (unqual != NULL) {
            scope_insert_alias(decl, unqual);

            if ((kind == T_CONST_DECL) || (kind == T_SIGNAL_DECL) || (kind == T_VAR_DECL))
               if (type_is_record(tree_type(decl)))
                  sem_declare_fields(tree_type(decl), unqual);
         }
      }
...
}
````

There are no confidences in this correction for me.
You might as well refactor code for record referring.

nickg, 2015-04-25:
Fixed now. I'll probably need to clean up how record fields work at some point.

closed 2015-04-25 by nickg