nickg/nvc #139: Configure problem with LLVM 3.6.0 under OS X

tmeissner, 2015-04-25:
When I try to run configure under OS X 10.10 with LLVM 3.6.0 installed under /opt//clang+llvm-3.6.0-x86_64-apple-darwin with the following command:

    $ ../configure --with-llvm=/opt/clang+llvm-3.6.0-x86_64-apple-darwin/bin/llvm-config

I get this error:

```
 ...
checking for LLVM (engine bitreader bitwriter)... no
checking for LLVM shared library... no
configure: error: We could not detect the llvm libraries make sure that llvm-config is on your path or specified by --with-llvm.
```

    $ grep -A 5 "checking for LLVM" config.log

shows this:

```
configure:9183: checking for LLVM (engine bitreader bitwriter)
configure:9205: g++ -o conftest -g -O2 -I/opt/clang+llvm-3.6.0-x86_64-apple-darwin/include  -DNDEBUG -D_GNU_SOURCE -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D__STDC_LIMIT_MACROS -O3  -std=c++11 -fvisibility-inlines-hidden -fno-exceptions -fno-rtti -fno-common -Wcast-qual   -L/opt/clang+llvm-3.6.0-x86_64-apple-darwin/lib  conftest.cpp -lz -ltcl  -lreadline -lLLVMBitWriter -lLLVMX86Disassembler -lLLVMX86AsmParser -lLLVMX86CodeGen -lLLVMSelectionDAG -lLLVMAsmPrinter -lLLVMCodeGen -lLLVMScalarOpts -lLLVMProfileData -lLLVMInstCombine -lLLVMTransformUtils -lLLVMipa -lLLVMAnalysis -lLLVMX86Desc -lLLVMMCDisassembler -lLLVMX86Info -lLLVMX86AsmPrinter -lLLVMX86Utils -lLLVMMCJIT -lLLVMTarget -lLLVMExecutionEngine -lLLVMRuntimeDyld -lLLVMObject -lLLVMMCParser -lLLVMBitReader -lLLVMMC -lLLVMCore -lLLVMSupport -L/opt/clang+llvm-3.6.0-x86_64-apple-darwin/lib  >&5
Undefined symbols for architecture x86_64:
  "_del_curterm", referenced from:
      llvm::sys::Process::FileDescriptorHasColors(int) in libLLVMSupport.a(Process.o)
  "_set_curterm", referenced from:
--
configure:9231: checking for LLVM shared library
configure:9259: g++ -o conftest -g -O2 -I/opt/clang+llvm-3.6.0-x86_64-apple-darwin/include  -DNDEBUG -D_GNU_SOURCE -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D__STDC_LIMIT_MACROS -O3  -std=c++11 -fvisibility-inlines-hidden -fno-exceptions -fno-rtti -fno-common -Wcast-qual   -L/opt/clang+llvm-3.6.0-x86_64-apple-darwin/lib  conftest.cpp -lz -ltcl  -lreadline -Wl,-rpath /opt/clang+llvm-3.6.0-x86_64-apple-darwin/lib -lLLVM-3.6.0 >&5
ld: library not found for -lLLVM-3.6.0
clang: error: linker command failed with exit code 1 (use -v to see invocation)
configure:9259: $? = 1
configure: program exited with status 1
```
The llvm-config tool shows these informations:
```
./llvm-config --ldflags
-L/opt/clang+llvm-3.6.0-x86_64-apple-darwin/lib

./llvm-config --libs
-lLLVMLTO -lLLVMObjCARCOpts -lLLVMLinker -lLLVMBitWriter -lLLVMIRReader -lLLVMAsmParser
-lLLVMR600CodeGen -lLLVMipo -lLLVMVectorize -lLLVMR600AsmParser -lLLVMR600Desc
-lLLVMR600Info -lLLVMR600AsmPrinter -lLLVMSystemZDisassembler -lLLVMSystemZCodeGen
-lLLVMSystemZAsmParser -lLLVMSystemZDesc -lLLVMSystemZInfo -lLLVMSystemZAsmPrinter
-lLLVMHexagonDisassembler -lLLVMHexagonCodeGen -lLLVMHexagonDesc -lLLVMHexagonInfo
-lLLVMNVPTXCodeGen -lLLVMNVPTXDesc -lLLVMNVPTXInfo -lLLVMNVPTXAsmPrinter
-lLLVMCppBackendCodeGen -lLLVMCppBackendInfo -lLLVMMSP430CodeGen -lLLVMMSP430Desc
-lLLVMMSP430Info -lLLVMMSP430AsmPrinter -lLLVMXCoreDisassembler -lLLVMXCoreCodeGen
-lLLVMXCoreDesc -lLLVMXCoreInfo -lLLVMXCoreAsmPrinter -lLLVMMipsDisassembler
-lLLVMMipsCodeGen -lLLVMMipsAsmParser -lLLVMMipsDesc -lLLVMMipsInfo -lLLVMMipsAsmPrinter
-lLLVMAArch64Disassembler -lLLVMAArch64CodeGen -lLLVMAArch64AsmParser -lLLVMAArch64Desc 
-lLLVMAArch64Info -lLLVMAArch64AsmPrinter -lLLVMAArch64Utils -lLLVMARMDisassembler
-lLLVMARMCodeGen -lLLVMARMAsmParser -lLLVMARMDesc -lLLVMARMInfo -lLLVMARMAsmPrinter
-lLLVMPowerPCDisassembler -lLLVMPowerPCCodeGen -lLLVMPowerPCAsmParser
-lLLVMPowerPCDesc -lLLVMPowerPCInfo -lLLVMPowerPCAsmPrinter -lLLVMSparcDisassembler
-lLLVMSparcCodeGen -lLLVMSparcAsmParser -lLLVMSparcDesc -lLLVMSparcInfo
-lLLVMSparcAsmPrinter -lLLVMTableGen -lLLVMDebugInfo -lLLVMOption -lLLVMX86Disassembler
-lLLVMX86AsmParser -lLLVMX86CodeGen -lLLVMSelectionDAG -lLLVMAsmPrinter -lLLVMX86Desc
-lLLVMMCDisassembler -lLLVMX86Info -lLLVMX86AsmPrinter -lLLVMX86Utils -lLLVMMCJIT
-lLLVMLineEditor -lLLVMInstrumentation -lLLVMInterpreter -lLLVMExecutionEngine -lLLVMRuntimeDyld
-lLLVMCodeGen -lLLVMScalarOpts -lLLVMProfileData -lLLVMObject -lLLVMMCParser
-lLLVMBitReader -lLLVMInstCombine -lLLVMTransformUtils -lLLVMipa -lLLVMAnalysis -lLLVMTarget
-lLLVMMC -lLLVMCore -lLLVMSupport
```

nickg, 2015-04-26:
This:

```
Undefined symbols for architecture x86_64:
  "_del_curterm", referenced from:
      llvm::sys::Process::FileDescriptorHasColors(int) in libLLVMSupport.a(Process.o)
  "_set_curterm", referenced from:
```

Looks like it's missing the curses library. Can you what `llvm-config --system-libs` outputs? The `--system-libs` option was added recently.

Regarding the second error about the shared library, can you see if there's any `*.so` or `*.dylib` in `/opt/clang+llvm-3.6.0-x86_64-apple-darwin/lib`? It's generally better to get the shared library version working as it reduces the executable size dramatically compared to static linking LLVM.

tmeissner, 2015-04-26:
Mhm, llvm-config thinks, that there is a curses library:

```
$ ./llvm-config --system-libs
-lz -lpthread -ledit -lcurses -lm
```

And a list of shared libs under lib:

```
$ find . -name "*.dylib"
./BugpointPasses.dylib
./clang/3.6.0/lib/darwin/libclang_rt.asan_iossim_dynamic.dylib
./clang/3.6.0/lib/darwin/libclang_rt.asan_osx_dynamic.dylib
./libclang.dylib
./libLTO.dylib
./LLVMHello.dylib
```
```
$ find . -name "*.so"
```

prints out nothing

nickg, 2015-04-26:
Can you try with the above commit? It should add the output of `llvm-config --system-libs` to the link command if linking against static LLVM >= 3.4.

tmeissner, 2015-04-26:
Yeah, great, now it works, thanks :)

But make check don't work:

```
$ make check
  CC       test/test_util.o
In file included from ../test/test_util.c:1:
../test/test_util.h:7:10: fatal error: 'check.h' file not found
#include <check.h>
         ^
1 error generated.
make[1]: *** [test/test_util.o] Error 1
make: *** [check-am] Error 2
```

 It seems it doesn't find the header file. But i found the right paths in the generated Makefile:

```
CHECK_CFLAGS = -D_THREAD_SAFE -I/usr/local/Cellar/check/0.9.14/include
CHECK_LIBS = -L/usr/local/Cellar/check/0.9.14/lib -lcheck
```

nickg, 2015-04-26:
Can you try `make check V=1`? That should show the arguments being passed to the compiler.

tmeissner, 2015-04-26:
Mhm, nothing to see of the CHECK_CFLAGS and CHECK_LIBS:

```
$ make check V=1
/Applications/Xcode.app/Contents/Developer/usr/bin/make  test/libtest_util.a bin/test_lib bin/test_ident bin/test_parse bin/test_sem bin/test_simp bin/test_elab bin/test_heap bin/test_hash bin/test_group bin/test_bounds bin/test_value bin/test_lower 
depbase=`echo test/test_util.o | sed 's|[^/]*$|.deps/&|;s|\.o$||'`;\
        gcc -DHAVE_CONFIG_H -I. -I..  -I../thirdparty -I../src  -Wall    -g -O2 -MT test/test_util.o -MD -MP -MF $depbase.Tpo -c -o test/test_util.o ../test/test_util.c &&\
        mv -f $depbase.Tpo $depbase.Po
In file included from ../test/test_util.c:1:
../test/test_util.h:7:10: fatal error: 'check.h' file not found
#include <check.h>
         ^
1 error generated.
make[1]: *** [test/test_util.o] Error 1
make: *** [check-am] Error 2
```

nickg, 2015-04-26:
The `CHECK_CFLAGS` variable wasn't being passed through. Try now?

tmeissner, 2015-04-26:
Yeah, make check runs now.
Test cover1 fails:
```
cover1 : failed (no match)
```
Maybe it can't find the testcase file?

nickg, 2015-04-26:
Can you post the contents of `logs/cover1/out`? Not sure why the output would be different though.

tmeissner, 2015-04-26:
```
$ cat logs/cover1/out 
/Users/torsten/Projects/Git/nvc/build/bin/nvc  -a /Users/torsten/Projects/Git/nvc/test/regress/cover1.vhd
/Users/torsten/Projects/Git/nvc/build/bin/nvc  -e cover1 --disable-opt --cover 
/bin/mv /Users/torsten/Projects/Git/nvc/build/logs/cover1/work/_WORK.COVER1.elab.bc /Users/torsten/Projects/Git/nvc/build/logs/cover1/work/_WORK.COVER1.final.bc
/Users/torsten/Projects/Git/nvc/build/bin/nvc  -r cover1
** Note: coverage report generated in /Users/torsten/Projects/Git/nvc/build/logs/cover1/work/WORK.COVER1.cover/
            0/11 statements covered
            0/0 branches covered
            0/0 conditions covered
```

The full error message for cover1 test is as follows:
```
cover1 : failed (no match)
    10/11 statements covered
```

nickg, 2015-04-26:
I managed to reproduce this on Linux: looks like the coverage feature does not work with LLVM 3.6 for some reason.

nickg, 2015-04-26:
Should be fixed by 945132f

closed 2015-04-26 by nickg

galderz, 2020-08-28:
> This:
>
> ```
> Undefined symbols for architecture x86_64:
>   "_del_curterm", referenced from:
>       llvm::sys::Process::FileDescriptorHasColors(int) in libLLVMSupport.a(Process.o)
>   "_set_curterm", [...]

Thx for tip! Helped me solved the same error when compiling the LLVM Kaleidoscope tutorial (see [here](https://gist.github.com/galderz/588cdb20c748250fe19a2872bf131e9f#file-gistfile1-txt-L28)).