nickg/nvc #146: Caught signal 11,  when refer to a field of a record type.

ikwzm, 2015-04-27:
````VHDL
-- a_ng.vhd
package A_NG is
    type      A_NG_TYPE is record
                  debug         : integer;        
    end record;
    procedure PROC_A(A_ARG:inout A_NG_TYPE);
end A_NG;
package body A_NG is
    procedure PROC_A(A_ARG:inout A_NG_TYPE) is
    begin
        null;
    end procedure;
    procedure PROC_B(B_ARG:inout A_NG_TYPE; B_VAL:out integer) is
        procedure PROC_C(C_VAL:out integer) is
        begin
            PROC_A(B_ARG);
            C_VAL := B_ARG.debug;
        end procedure;
    begin
        PROC_C(B_VAL);
    end procedure;
end A_NG;
````

````
ikwzm@vm-ubuntu:~/work/nvc/issue-11-test$ ../build/bin/nvc -L ../build/lib -a a_ng.vhd

*** Caught signal 11 (SIGSEGV) [address=0x8, ip=0x7f70697fc530] ***

-------- STACK TRACE --------
/usr/lib/x86_64-linux-gnu/libLLVM-3.4.so.1(_ZNK4llvm4Type13getScalarTypeEv+0) [0x7f70697fc530]
/usr/lib/x86_64-linux-gnu/libLLVM-3.4.so.1(_ZNK4llvm4Type13getScalarTypeEv+0) [0x7f70697fc530]
/usr/lib/x86_64-linux-gnu/libLLVM-3.4.so.1(_ZNK4llvm4Type22getPointerAddressSpaceEv+0x9) [0x7f70697fc8c9]
/usr/lib/x86_64-linux-gnu/libLLVM-3.4.so.1(LLVMBuildStructGEP+0xb8) [0x7f706975ef68]
../build/bin/nvc() [0x46b59b]
../build/bin/nvc() [0x46f90a]
../build/bin/nvc() [0x471199]
../build/bin/nvc() [0x4716fa]
../build/bin/nvc() [0x471328]
../build/bin/nvc(cgen+0x5f5) [0x472125]
../build/bin/nvc(main+0x1263) [0x411d93]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf5) [0x7f7067c72ec5]
../build/bin/nvc() [0x412045]
-----------------------------
````

ikwzm, 2015-04-27:
cgen_op_record_ref() at cgen.c
````C
static void cgen_op_record_ref(int op, cgen_ctx_t *ctx)
{
   vcode_reg_t result = vcode_get_result(op);
   ctx->regs[result] = LLVMBuildStructGEP(builder, cgen_get_arg(op, 0, ctx),
                                          vcode_get_field(op),
                                          cgen_reg_name(result));

   LLVMTypeRef field_type = LLVMGetElementType(LLVMTypeOf(ctx->regs[result]));
   if (LLVMGetTypeKind(field_type) == LLVMArrayTypeKind) {
      LLVMValueRef indexes[] = {
         llvm_int32(0),
         llvm_int32(0)
      };
      ctx->regs[result] = LLVMBuildGEP(builder, ctx->regs[result], indexes,
                                       ARRAY_LEN(indexes), "");
   }
}
````

Caught signal 11 (SIGSEGV) at LLVMBuildStructGEP().

It was analyzed to here, but unfortunately, I don't understand well about LLVM...

nickg, 2015-04-27:
Unfortunately there was a bug in the nested subprogram code where it counted parameters only for functions not procedures.

closed 2015-04-27 by nickg