nickg/nvc #153: two bugs within generate

postoroniy, 2015-05-01:
1) nvc should failed when there's access out of boundary array inside of generate loop, (ghdl/modelsim do that)
2) nvc drives the same value on inputs of all instances inside generate loop when there's condition drive such a value just for first instance
Please have look source below:

entity test_inst is
generic(
    G_ROUND         : natural  := 0;
    G_ROUND_ENABLE  : boolean  := false
);
port(
    i_value         : in bit_vector(7 downto 0);
    o_ena           : out bit;
    o_value         : out bit_vector(7 downto 0)
);
end test_inst;
architecture rtl of test_inst is
begin
    o_ena <='1' when G_ROUND_ENABLE else '0';
    o_value <=(others=>'1') when G_ROUND=1 and G_ROUND_ENABLE else not i_value;
end architecture rtl;
entity tb is
end entity tb;
architecture beh of tb is
    constant G_ROUND_ENABLE:boolean:=true;
    constant C_ADDROUND : bit_vector(7 downto 0):="00001111";
    constant C_ZERO8 : bit_vector(7 downto 0):=(others=>'0');
    signal s_ena:bit_vector(7 downto 0);
    type T_IN_DATA  is array(integer range<>) of bit_vector(7 downto 0);
    --signal s_value: T_IN_DATA(7 downto -1);-- this should work anyway
    signal s_value: T_IN_DATA(7 downto 0);--this is for bug 1, nvc should report error
begin

    GEN_MACS_V : for v in 0 to 7 generate
        signal C   :bit_vector(7 downto 0);
        signal D   :bit_vector(7 downto 0);
    begin
        --should fail here, but doesn't
        --GHDL failed here with "bound check failure"
        -- ghdl drives correct values on each instances, nvc doesn't
        C    <= C_ADDROUND when v=0 and G_ROUND_ENABLE else s_value(v-1);--bug 1
        -- below is workaround, but I am lazy enough to not use it :))))
        --c_gen: if v=0 and G_ROUND_ENABLE generate
        --    C    <= C_ADDROUND;
        --end generate c_gen;
        --nc_gen: if v>0  generate
        --    C <= s_value(v-1);
        --end generate nc_gen; 

        test_i : entity work.test_inst
        generic map(
            G_ROUND         => 1
        )
        port map(
            i_value     => C,
            o_ena       => s_ena(v),
            o_value     => s_value(v)
        );     
    end generate GEN_MACS_V;

    process
    begin
        wait for 1 ns;
        assert s_value(0) = not C_ADDROUND;
        assert s_value(1) = C_ADDROUND;
    end process;
end architecture;

nickg, 2015-05-03:
NVC does give an error if you access an array out of bounds. E.g. if you change your code to:

```vhdl
C <=  s_value(v-1);--bug 1
```

Then it fails at runtime with:

```
** Fatal: array index -1 outside bounds 7 downto 0
        File ../test/regress/issue153.vhd, Line 39
```

There's perhaps a bug here that the error is not caught at elaboration time, but your original example is guarded against `v = 0` so should not produce an error:

```vhdl
C    <= C_ADDROUND when v=0 and G_ROUND_ENABLE else s_value(v-1);--bug 1
```

If you look in the 2008 LRM, section 11.6 specifies a translation of concurrent signal assignment statements to a process with an `if` statement. Section 10.8 specifies the behaviour of `if` statements as

> If one condition evaluates to TRUE, then the corresponding sequence of statements is executed; otherwise, none of the sequences of statements is executed.

So `s_value(v-1)` is not executed in the case where `v=0`.

nickg, 2015-05-03:
The simulation mis-match is fixed now. If you set `G_ROUND_ENABLE` to `false` to it evaluates `s_value(v-1)` you'll get an error at elaboration time now:

```
** Error: array S_VALUE index -1 out of bounds 7 downto 0
        File ../test/regress/issue153.vhd, Line 39
```

closed 2015-05-03 by nickg

postoroniy, 2015-05-03:
First of all I checked those things with ghdl and modelsim. They complained about out of bond access. 
I do understand that it sounds quite stupid here-> C    <= C_ADDROUND when v=0 and G_ROUND_ENABLE else s_value(v-1); 
And this statement is not "if" statement because it is inside of generate loop and variable V is not available during simulation as well as  that it should be converted to "appropriate circuitry". It means there's no mux at all.
Also, this bug 1 is just side effect of mine debugging process around bug 2. anyway bug 1 doesn't bite me :).
But not bug 2.  It seem that nvc dropped all statement  "when v=0 and G_ROUND_ENABLE" simply taking C_ADDROUND as a result of evaluation.

nickg, 2015-05-03:
The LRM doesn't say anything about what sort of circuit you'd get from synthesising the code. All it specifies for concurrent signal assignment statements is how to transform them into an equivalent process containing an `if`-statement.

postoroniy, 2015-05-03:
There's still bug 2 anyway

nickg, 2015-05-03:
Can you elaborate a bit? The two assertions in your example no longer fail with commit a13ac75 above.

postoroniy, 2015-05-03:
Oops, sorry, I didn't install.
All good!

nickg, 2015-05-03:
OK, great.