nickg/nvc #154: build of XilinxCoreLib crashes

gecomaia, 2015-05-02:
nvc --work=/home/ed/.nvc/lib/xilinxcorelib -a --relax=prefer-explicit /opt/xilinx/ise147/14.7/ISE_DS/ISE/vhdl/src/XilinxCoreLib/mult_gen_pkg_v11_2.vhd
nvc: ../src/tree.c:956: tree_name: Assertion `item->tree != ((void *)0)' failed.

*** Caught signal 6 (SIGABRT) ***

-------- STACK TRACE --------
/lib/x86_64-linux-gnu/libc.so.6(gsignal+0x39) [0x7f61e3ddbcc9]
/lib/x86_64-linux-gnu/libc.so.6(gsignal+0x39) [0x7f61e3ddbcc9]
/lib/x86_64-linux-gnu/libc.so.6(abort+0x148) [0x7f61e3ddf0d8]
/lib/x86_64-linux-gnu/libc.so.6(+0x2fb86) [0x7f61e3dd4b86]
/lib/x86_64-linux-gnu/libc.so.6(+0x2fc32) [0x7f61e3dd4c32]
nvc(tree_name+0x75) [0x42eb05]
nvc() [0x45d63d]
nvc() [0x45e5e9]
nvc(lower_unit+0xb6) [0x45f486]
nvc(main+0x1255) [0x411dd5]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf5) [0x7f61e3dc6ec5]
nvc() [0x412095]

postoroniy, 2015-05-04:
I tried as well and mine is not crashing, but I have strange error
./tools/build-xilinx.rb 
Using ISE installation in /usr/Xilinx/14.7/ISE_DS/ISE

------ UNISIM package ------
nvc --work=/home/slava/.nvc/lib/unisim -a --relax=prefer-explicit /usr/Xilinx/14.7/ISE_DS/ISE/vhdl/src/unisims/unisim_VPKG.vhd
nvc --work=/home/slava/.nvc/lib/unisim -a --relax=prefer-explicit /usr/Xilinx/14.7/ISE_DS/ISE/vhdl/src/unisims/unisim_VCOMP.vhd
nvc --work=/home/slava/.nvc/lib/unisim --codegen vpkg
nvc --work=/home/slava/.nvc/lib/unisim --codegen vcomponents

------ Primitives ------
nvc --work=/home/slava/.nvc/lib/unisim -a --relax=prefer-explicit /usr/Xilinx/14.7/ISE_DS/ISE/vhdl/src/unisims/primitive/AND2B1L.vhd
nvc --work=/home/slava/.nvc/lib/unisim -a --relax=prefer-explicit /usr/Xilinx/14.7/ISE_DS/ISE/vhdl/src/unisims/primitive/AUTOBUF.vhd
nvc --work=/home/slava/.nvc/lib/unisim -a --relax=prefer-explicit /usr/Xilinx/14.7/ISE_DS/ISE/vhdl/src/unisims/primitive/BSCAN_VIRTEX6.vhd
nvc --work=/home/slava/.nvc/lib/unisim -a --relax=prefer-explicit /usr/Xilinx/14.7/ISE_DS/ISE/vhdl/src/unisims/primitive/BUFH.vhd
nvc --work=/home/slava/.nvc/lib/unisim -a --relax=prefer-explicit /usr/Xilinx/14.7/ISE_DS/ISE/vhdl/src/unisims/primitive/BUFHCE.vhd
nvc --work=/home/slava/.nvc/lib/unisim -a --relax=prefer-explicit /usr/Xilinx/14.7/ISE_DS/ISE/vhdl/src/unisims/primitive/BUFIODQS.vhd
nvc --work=/home/slava/.nvc/lib/unisim -a --relax=prefer-explicit /usr/Xilinx/14.7/ISE_DS/ISE/vhdl/src/unisims/primitive/CAPTURE_VIRTEX6.vhd
nvc --work=/home/slava/.nvc/lib/unisim -a --relax=prefer-explicit /usr/Xilinx/14.7/ISE_DS/ISE/vhdl/src/unisims/primitive/DSP48E1.vhd
nvc --work=/home/slava/.nvc/lib/unisim -a --relax=prefer-explicit /usr/Xilinx/14.7/ISE_DS/ISE/vhdl/src/unisims/primitive/EFUSE_USR.vhd
nvc --work=/home/slava/.nvc/lib/unisim -a --relax=prefer-explicit /usr/Xilinx/14.7/ISE_DS/ISE/vhdl/src/unisims/primitive/FIFO18E1.vhd
nvc --work=/home/slava/.nvc/lib/unisim -a --relax=prefer-explicit /usr/Xilinx/14.7/ISE_DS/ISE/vhdl/src/unisims/primitive/FIFO36E1.vhd
nvc --work=/home/slava/.nvc/lib/unisim -a --relax=prefer-explicit /usr/Xilinx/14.7/ISE_DS/ISE/vhdl/src/unisims/primitive/FRAME_ECC_VIRTEX6.vhd
nvc --work=/home/slava/.nvc/lib/unisim -a --relax=prefer-explicit /usr/Xilinx/14.7/ISE_DS/ISE/vhdl/src/unisims/primitive/IBUFDS_GTXE1.vhd
nvc --work=/home/slava/.nvc/lib/unisim -a --relax=prefer-explicit /usr/Xilinx/14.7/ISE_DS/ISE/vhdl/src/unisims/primitive/ICAP_VIRTEX6.vhd
nvc --work=/home/slava/.nvc/lib/unisim -a --relax=prefer-explicit /usr/Xilinx/14.7/ISE_DS/ISE/vhdl/src/unisims/primitive/IOBUFDS_DIFF_OUT.vhd
nvc --work=/home/slava/.nvc/lib/unisim -a --relax=prefer-explicit /usr/Xilinx/14.7/ISE_DS/ISE/vhdl/src/unisims/primitive/IODELAYE1.vhd
nvc --work=/home/slava/.nvc/lib/unisim -a --relax=prefer-explicit /usr/Xilinx/14.7/ISE_DS/ISE/vhdl/src/unisims/primitive/ISERDESE1.vhd
nvc --work=/home/slava/.nvc/lib/unisim -a --relax=prefer-explicit /usr/Xilinx/14.7/ISE_DS/ISE/vhdl/src/unisims/primitive/JTAG_SIM_VIRTEX6.vhd
nvc --work=/home/slava/.nvc/lib/unisim -a --relax=prefer-explicit /usr/Xilinx/14.7/ISE_DS/ISE/vhdl/src/unisims/primitive/MMCM_ADV.vhd
** Error: ambiguous call to function SLV_TO_INT
              SLV_TO_INT [STD_LOGIC_VECTOR return INTEGER]
              SLV_TO_INT [STD_LOGIC_VECTOR return INTEGER]
	File /usr/Xilinx/14.7/ISE_DS/ISE/vhdl/src/unisims/primitive/MMCM_ADV.vhd, Line 461
               clk_ht1 := 2 * SLV_TO_INT(clk_ht) + 1;
                              ^^^^^^^^^^^^^^^^^^
** Error: ambiguous call to function SLV_TO_INT
              SLV_TO_INT [STD_LOGIC_VECTOR return INTEGER]
              SLV_TO_INT [STD_LOGIC_VECTOR return INTEGER]
	File /usr/Xilinx/14.7/ISE_DS/ISE/vhdl/src/unisims/primitive/MMCM_ADV.vhd, Line 465
           clk_div_tmp := SLV_TO_INT(clk_ht) + SLV_TO_INT(clk_lt);

I can't add any extra info( about altera/lattice).

nickg, 2015-05-04:
The second problem mentioned by @postoroniy is fixed by 0efd9ce. The `unisim` library now builds for me (at least with the old 13.4 install I have). I don't have 14.7 on this PC, and my `mult_gen_pkg_v11_2.vhd` fails with a different error. I'll try tomorrow with a more recent ISE version.

postoroniy, 2015-05-04:
unisim 14.7 is built with no error now.

gecomaia, 2015-05-07:
I'm sorry for the late reply, 
just tried to compile XilinxCoreLibs and fails with the following error:

nvc --work=/home/ed/.nvc/lib/xilinxcorelib -a --relax=prefer-explicit /opt/xilinx/ise147/14.7/ISE_DS/ISE/vhdl/src/XilinxCoreLib/mult_gen_pkg_v11_2.vhd
** Error: unexpected use while parsing package body declarative item, expecting
          one of function, procedure, shared, impure, pure, attribute, type,
          constant, subtype or alias
	File /opt/xilinx/ise147/14.7/ISE_DS/ISE/vhdl/src/XilinxCoreLib/mult_gen_pkg_v11_2.vhd, Line 3039
      use ieee.std_logic_signed.all;
      ^^^

ps: my machine is : Ubuntu 14.04.2 LTS, x86_64

nickg, 2015-05-09:
I've pushed a bunch of changes that fix most of the errors building XilinxCoreLib. There's still one bug that prevents us analysing the `fir_compiler_*.vhd` files, and the `mult_gen_*.vhd` files still fail with this error:

```
** Error: no visible declaration for IEEE.STD_LOGIC_UNSIGNED."*"
        File /opt/Xilinx/13.4/ISE_DS/ISE/vhdl/src/XilinxCoreLib/mult_gen_v11_2.vhd, Line 329
            pm <= std_logic_vector(RESIZE(unsigned(ieee.std_logic_unsigned."*"(ai,  ...
                                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
```

As far as I can tell this is a genuine error in the Xilinx code (although ModelSim accepts it): it's missing a use clause for `ieee.std_logic_unsigned`. If you need this particular component then add the following to the top of the file and remove it from `xcl_skip` in `build-xilinx.rb`. 

```vhdl
use ieee.std_logic_unsigned;   -- No .all
```

Otherwise that script is now able to build all of Unisim and XilinxCoreLib. I haven't actually tried running any simulations that use it: I'd be interested to hear how you get on...

gecomaia, 2015-05-09:
Great work!

I've added some more files into the skip section, then all compiles fine.
'floating_point_pkg_v6_2.vhd',
'floating_point_pkg_v6_1.vhd',
# Dependency of above
'floating_point_v6_2.vhd',
'floating_point_v6_1.vhd',

But when I try to elaborate a small sub-unit, I get the following errors
nvc -a *.vhd ........ everything ok, no issues
nvc -e ip_amu_tb

** Error: cannot find entity for component OBUFDS without binding indication
	File ip_amu_serpar.vhd, Line 313
** Error: cannot find entity for component IBUFDS without binding indication
	File ip_amu_serpar.vhd, Line 339
** Error: cannot find entity for component BUFR without binding indication
	File ip_amu_serpar.vhd, Line 348
....... (and so on)

for example BUFR:

Declaration:
==========
component BUFR
  generic ( BUFR_DIVIDE                : string := "BYPASS";
            SIM_DEVICE                 : string := "7SERIES");
  port (    O                          : out STD_LOGIC;
            CE                         : in  STD_LOGIC;
            CLR                        : in  STD_LOGIC;
            I                          : in  STD_LOGIC);
end component;
attribute BOX_TYPE of BUFR : component is "PRIMITIVE";

Instantiation:
==========

      BUF_DATA_CLK : BUFR  generic map (  BUFR_DIVIDE    => "BYPASS",               -- Values: "BYPASS, 1, 2, 3, 4, 5, 6, 7, 8"
                                          SIM_DEVICE     => "7SERIES")              -- Must be set to "7SERIES"
                              port map (  O              => amu_adc_dco,            -- 1-bit output: Clock output port
                                          CE             => '1',                    -- 1-bit input: Active high, clock enable (Divided modes only)
                                          CLR            => '0',                    -- 1-bit input: Active high, asynchronous clear (Divided modes only)
                                          I              => amu_adc_dco_i);         -- 1-bit input: Clock buffer input driven by an IBUFG, MMCM or local interconnect

What can I do next?
lg

nickg, 2015-05-13:
Sorry for the delay... is there a `library unisim;` clause for the unit in `ip_amu_serpar.vhd`?

gecomaia, 2015-05-14:
yes, I include the following libs:

     library IEEE;
     use IEEE.std_logic_1164.all;
     use IEEE.std_logic_arith.all;
     use IEEE.std_logic_misc.all;
     use IEEE.std_logic_unsigned.all;
     use IEEE.math_real.all;
     use WORK.MAIN_PACK.all;
     library unisim;
     use unisim.vcomponents.all;
     use UNISIM.all;

     -- synopsys translate_off
     Library XilinxCoreLib;
     -- synopsys translate_on

nickg, 2015-05-15:
I'm not able to reproduce this. Can you try the example below?

```vhdl
entity bufr_test is
end entity;

library ieee;
use ieee.std_logic_1164.all;

library unisim;
use unisim.vcomponents.all;

architecture test of bufr_test is
    component BUFR
        generic ( BUFR_DIVIDE : string := "BYPASS";
                  SIM_DEVICE  : string := "7SERIES");
        port (    O           : out STD_LOGIC;
                  CE          : in  STD_LOGIC;
                  CLR         : in  STD_LOGIC;
                  I           : in  STD_LOGIC);
    end component;
    attribute BOX_TYPE of BUFR : component is "PRIMITIVE";

    signal amu_adc_dco_i : std_logic;
    signal amu_adc_dco   : std_logic;
begin

    BUF_DATA_CLK : BUFR
        generic map (  BUFR_DIVIDE    => "BYPASS",
                       SIM_DEVICE     => "7SERIES")
        port map (  O              => amu_adc_dco,
                    CE             => '1',
                    CLR            => '0',
                    I              => amu_adc_dco_i);

end architecture;
```

gecomaia, 2015-05-17:
I tried your example - and it worked;
so I scaled down my module and tried to find out, why my module didn't work.
 
If you move the **entity** declaration below the library include section then you get
the error:

    ** Error: cannot find entity for component BUFR without binding indication
	        File bufr_test.vhd, Line 26

```vhdl
library ieee;
use ieee.std_logic_1164.all;

library unisim;
use unisim.vcomponents.all;

entity bufr_test is
end entity;

architecture test of bufr_test is
    component BUFR
        generic ( BUFR_DIVIDE : string := "BYPASS";
                  SIM_DEVICE  : string := "7SERIES");
        port (    O           : out STD_LOGIC;
                  CE          : in  STD_LOGIC;
                  CLR         : in  STD_LOGIC;
                  I           : in  STD_LOGIC);
    end component;
    attribute BOX_TYPE of BUFR : component is "PRIMITIVE";

    signal amu_adc_dco_i : std_logic;
    signal amu_adc_dco   : std_logic;
begin

    BUF_DATA_CLK : BUFR
.....
```

nickg, 2015-05-17:
Thanks. Can you try now with the above commit?

gecomaia, 2015-05-17:
Thank you,
the issue with the missing components is solved, but when I try to elaborate the whole module, I get
the following error message:

```
nvc: ../src/common.c:54: range_bounds: Assertion `folded' failed.

*** Caught signal 6 (SIGABRT) ***

-------- STACK TRACE --------
/lib/x86_64-linux-gnu/libc.so.6(gsignal+0x39) [0x7fc05b10dcc9]
/lib/x86_64-linux-gnu/libc.so.6(gsignal+0x39) [0x7fc05b10dcc9]
/lib/x86_64-linux-gnu/libc.so.6(abort+0x148) [0x7fc05b1110d8]
/lib/x86_64-linux-gnu/libc.so.6(+0x2fb86) [0x7fc05b106b86]
/lib/x86_64-linux-gnu/libc.so.6(+0x2fc32) [0x7fc05b106c32]
nvc() [0x44d0f8]
nvc(type_width+0x11a) [0x4316fa]
nvc() [0x443dc8]
nvc() [0x443f11]
nvc() [0x444fb1]
nvc() [0x444a2a]
nvc() [0x444fbc]
nvc() [0x444a2a]
nvc() [0x4445cd]
nvc() [0x444fbc]
-----------------------------
```

Doing the eleboration on one of the submoduls (IP_**AMU**_SERPAR == Angle Measurement Unit),
I get the following FATAL :

```
$ nvc -e IP_AMU_SERPAR
** Fatal: tree kind T_RECORD_REF not yet supported for offset calculation
	File ip_amu_serpar.vhd, Line 688

```

the according definition is:

```vhdl

......
entity IP_AMU_SERPAR is
    generic ( CONST : AMU_SERPAR_CONST_TYPE := AMU_SERPAR_CONST);
    port (
--------------------------------------------------------------------------------------+-----------------------------------------
        GSR:                          in  STD_LOGIC;                                --|  Global SET/RESET
        clk_adc_sample:               in  STD_LOGIC;                                --|  clock
        clk_sync:                     in  STD_LOGIC;                                --|  clock
--------------------------------------------------------------------------------------+-----------------------------------------
        AMU_SERPAR_IN:                in  AMU_SERPAR_IN_TYPE;                       --|
        AMU_SERPAR_OUT:               out AMU_SERPAR_OUT_TYPE;                      --|
--------------------------------------------------------------------------------------+-----------------------------------------
        dcm_rst:                      in  STD_LOGIC;                                --| Reset for DCM_Lock
        dcm_locked:                   out STD_LOGIC;                                --|
        TEST:                         out STD_LOGIC_VECTOR(31 downto 0)             --|
--------------------------------------------------------------------------------------+-----------------------------------------
    );
end IP_AMU_SERPAR;

......

   ADC_CHANNELS_OUTPUT_GEN : for I in 0 to AMU_SERPAR_NR_OF_CHANNELS-1 generate
      AMU_SERPAR_OUT.DATA_OUT(I)(9 downto 0) <= AMU_ADC_r(I)(9 downto 0);
   end generate ADC_CHANNELS_OUTPUT_GEN;

......

```

and the definition of AMU_SERPAR_??_TYPE is done in a package file (here is the snippet):

```vhdl

   type AMU_SERPAR_IN_TYPE is
      record
         DBG_X                         : STD_LOGIC_VECTOR(10 downto 0);
         DBG_Y                         : STD_LOGIC_VECTOR(10 downto 0);
         dbg_nd                        : STD_LOGIC;
         DATA_P                        : STD_LOGIC_VECTOR(AMU_SERPAR_NR_OF_CHANNELS-1 downto 0);
         DATA_N                        : STD_LOGIC_VECTOR(AMU_SERPAR_NR_OF_CHANNELS-1 downto 0);
         FCO_P                         : STD_LOGIC;
         FCO_N                         : STD_LOGIC;
         DCO_P                         : STD_LOGIC;
         DCO_N                         : STD_LOGIC;
      end record;

   type AMU_SERPAR_DATA_OUT_TYPE       is array(AMU_SERPAR_NR_OF_CHANNELS-1 downto 0) of STD_LOGIC_VECTOR(9 downto 0);

   type AMU_SERPAR_OUT_TYPE is
      record
         DBG_P                         : STD_LOGIC_VECTOR(10 downto 0);
         dbg_rdy                       : STD_LOGIC;
         dco                           : STD_LOGIC;
         CLK_P                         : STD_LOGIC;
         CLK_N                         : STD_LOGIC;
         DATA_OUT                      : AMU_SERPAR_DATA_OUT_TYPE;
         data_out_rdy                  : STD_LOGIC;
      end record;

   type IP_AMU_DBG_VALUE_TYPE          is array(AMU_SERPAR_NR_OF_CHANNELS-1 downto 0) of STD_LOGIC_VECTOR(9 downto 0);

```

hope this helps to isolate this behavior
lg
Gerhard

postoroniy, 2015-05-18:
xilinxcorelib build failed with errors:
....
nvc --work=/home/slava/.nvc/lib/xilinxcorelib -a --relax=prefer-explicit /usr/Xilinx/14.7/ISE_DS/ISE/vhdl/src/XilinxCoreLib/global_util_pkg_v1_1.vhd
** Warning: statement is unreachable
	File /usr/Xilinx/14.7/ISE_DS/ISE/vhdl/src/XilinxCoreLib/global_util_pkg_v1_1.vhd, Line 417
        return false;
        ^^^^^^^^^^^^^
nvc --work=/home/slava/.nvc/lib/xilinxcorelib -a --relax=prefer-explicit /usr/Xilinx/14.7/ISE_DS/ISE/vhdl/src/XilinxCoreLib/axi_utils_v1_1_comps.vhd
nvc --work=/home/slava/.nvc/lib/xilinxcorelib -a --relax=prefer-explicit /usr/Xilinx/14.7/ISE_DS/ISE/vhdl/src/XilinxCoreLib/mult_gen_pkg_v11_2.vhd
nvc --work=/home/slava/.nvc/lib/xilinxcorelib -a --relax=prefer-explicit /usr/Xilinx/14.7/ISE_DS/ISE/vhdl/src/XilinxCoreLib/floating_point_v6_2_consts.vhd
nvc --work=/home/slava/.nvc/lib/xilinxcorelib -a --relax=prefer-explicit /usr/Xilinx/14.7/ISE_DS/ISE/vhdl/src/XilinxCoreLib/floating_point_pkg_v6_2.vhd
** Error: ambiguous use of enumeration literal B_FLT_TO_FIX
	File /usr/Xilinx/14.7/ISE_DS/ISE/vhdl/src/XilinxCoreLib/floating_point_pkg_v6_2.vhd, Line 8252
        for i in b_flt_to_fix to b_Xf loop
                 ^^^^^^^^^^^^
** Error: ambiguous use of enumeration literal B_RECOMB
	File /usr/Xilinx/14.7/ISE_DS/ISE/vhdl/src/XilinxCoreLib/floating_point_pkg_v6_2.vhd, Line 8259
        for i in b_e2zm1 to b_recomb loop
                            ^^^^^^^^
** Error: ambiguous use of enumeration literal B_RECOMB
	File /usr/Xilinx/14.7/ISE_DS/ISE/vhdl/src/XilinxCoreLib/floating_point_pkg_v6_2.vhd, Line 8423
        for block_i in b_e2zm1 to b_recomb loop
                                  ^^^^^^^^
** Error: ambiguous use of enumeration literal B_FLT_TO_FIX
	File /usr/Xilinx/14.7/ISE_DS/ISE/vhdl/src/XilinxCoreLib/floating_point_pkg_v6_2.vhd, Line 9246
        for block_i in b_flt_to_fix to b_result_mag loop

nickg, 2016-07-24:
I'm going to close this now as it's been open for some time and the original crash was resolved. I have an old version of ISE 13.4 where corelib can be compiled without error now.

closed 2016-07-24 by nickg