nickg/nvc #157: When using the component declared in the package, when elaboration, it isn't possible to find it.

ikwzm, 2015-05-05:
````VHDL
-- file a_ng.vhd
package COMPONENTS is
    component DUMMY_MODULE
        port (I : in bit; O : out bit);
    end component;
end package;

use     WORK.COMPONENTS.DUMMY_MODULE;
entity  DUMMY_TOP is
    port (I : in bit; O : out bit);
end entity;
architecture RTL of DUMMY_TOP is
begin
    U: DUMMY_MODULE port map(I=>I, O=>O);
end architecture;

entity DUMMY_MODULE is
    port (I: in bit; O: out bit);
end entity;
architecture RTL of DUMMY_MODULE is
begin
    O <= I;
end architecture;
````

````
ikwzm@vm-ubuntu:~/work/nvc/test$ ../build/bin/nvc -L ../build/lib -L ./ -a a_ng.vhd
ikwzm@vm-ubuntu:~/work/nvc/test$ ../build/bin/nvc -L ../build/lib -L ./ -e DUMMY_TOP
** Error: cannot find entity for component WORK.COMPONENTS.DUMMY_MODULE without
          binding indication
	File a_ng.vhd, Line 14
````

ikwzm, 2015-05-05:
It’s me again!
Thank you for your time.

elab_default_binding() at elab.c

````C
static tree_t elab_default_binding(tree_t inst, lib_t *new_lib,
                                   const elab_ctx_t *ctx)
{
   // Default binding indication is described in LRM 93 section 5.2.2

   tree_t comp = tree_ref(inst);

   ident_t full_i = tree_ident(comp);
   ident_t lib_i = ident_until(full_i, '.');

   lib_t lib = NULL;
   bool search_others = true;
   if (lib_i == full_i) {
      lib    = lib_work();
      full_i = ident_prefix(lib_name(lib), full_i, '.');
   }
   else {
      search_others = false;
      if ((lib = lib_find(istr(lib_i), true, true)) == NULL)
         return NULL;
   }
...
}
````

But, if full_i is the form like LIBRARY_NAME.PACKAGE_NAME.COMPONENT_NAME,  isn't the entity we'd like to bind COMPONENT_NAME in LIBRARY_NAME ? 
So how about such one?

````C
static tree_t elab_default_binding(tree_t inst, lib_t *new_lib,
                                   const elab_ctx_t *ctx)
{
   // Default binding indication is described in LRM 93 section 5.2.2

   tree_t comp = tree_ref(inst);

   ident_t full_i = tree_ident(comp);
   ident_t lib_i = ident_until(full_i, '.');

   lib_t lib = NULL;
   bool search_others = true;
   if (lib_i == full_i) {
      lib    = lib_work();
      full_i = ident_prefix(lib_name(lib), full_i, '.');
   }
   else {
      search_others = false;
      if ((lib = lib_find(istr(lib_i), true, true)) == NULL)
         return NULL;
      full_i = ident_prefix(lib_name(lib), ident_rfrom(full_i, '.'), '.');  ////// <- How about such one?
   }
...
}
````

nickg, 2015-05-05:
Yes that looks right. Thanks!

closed 2015-05-05 by nickg