nickg/nvc #159: When using the component declared in the package of library except for work, when elaboration, it isn't possible to find it.

ikwzm, 2015-05-05:
````VHDL
-- file dummy.vhd
package COMPONENTS is
    component DUMMY_MODULE
        port (I : in bit; O : out bit);
    end component;
end package;

entity DUMMY_MODULE is
    port (I: in bit; O: out bit);
end entity;
architecture RTL of DUMMY_MODULE is
begin
    O <= I;
end architecture;
````

````VHDL
-- file dummy_top.vhd
use     DUMMY.COMPONENTS.DUMMY_MODULE;
entity  DUMMY_TOP is
    port (I : in bit; O : out bit);
end entity;
architecture RTL of DUMMY_TOP is
begin
    U: DUMMY_MODULE port map(I=>I, O=>O);
end architecture;
````

````
ikwzm@vm-ubuntu:~/work/nvc/issue-15-test$ ../build/bin/nvc -L ../build/lib -L ./ --work=dummy -a dummy.vhd
ikwzm@vm-ubuntu:~/work/nvc/issue-15-test$ ../build/bin/nvc -L ../build/lib -L ./ -a dummy_top.vhd
ikwzm@vm-ubuntu:~/work/nvc/issue-15-test$ ../build/bin/nvc -L ../build/lib -L ./ -e dummy_top
** Error: cannot find entity for component DUMMY.COMPONENTS.DUMMY_MODULE
          without binding indication
	File dummy_top.vhd, Line 8
````

ikwzm, 2015-05-06:
Thank you for your time.
After issue #157, a similar problem.

elab_default_binding() at elab.c

````C
static tree_t elab_default_binding(tree_t inst, lib_t *new_lib,
                                   const elab_ctx_t *ctx)
{
...
   tree_t entity = NULL;
   lib_search_params_t params = { lib_work(), full_i, &entity };
   lib_walk_index(params.lib, find_entity, &params);
...
}
````

Is it right to do as follows?

````C
   tree_t entity = NULL;
   lib_search_params_t params = { lib, full_i, &entity };  //// lib_work() -> lib
   lib_walk_index(params.lib, find_entity, &params);
````

nickg, 2015-05-06:
Thanks, fixed now.

closed 2015-05-06 by nickg