nickg/nvc #161: Feature request: Separate library logical name from library location in command line interface.

kraigher, 2015-05-06:
As I understand it the command line interface it currently only supports providing a logical name to the `--work` argument. A folder with the same name in the current working directory is created. In some verification scripts you would also want to control the location of the library folder separately from the library logical name. Thus one can manage the placement of the libraries for instance inside a temporary output folder.

Some simulators support having a configuration file which lets you specify:
```
library_logical_name = library_path
``` 
Typically a compile script would then first create all the library folders and map them to their logical name. After this the individual files would be compiled into their respective libraries with depenencies in other libraries found through the mapping.

Maybe it would be enough just have the `--work` flag support a `--work=logical_name:path` syntax.

nickg, 2015-05-09:
This is partially possible with the current `--work` option: you can specify a directory path and the last component will be the logical name. For example `--work=/x/y/z` the library name will be 'Z' and a directory will be created under `/x/y`. What you can't do at the moment is change the name of the physical library directory.

kraigher, 2015-05-09:
@nickg I see. That is more workable but leaves me with a few worries, some with the `--work` flag as you describe it and some with the `-L` flag. I might be biased from using other simulators but let me explain my perspective on this. In VUnit we expose a Python interface for users to define their library hierarchy. We expose two methods for adding libraries:

1. `add_library(library_name)`
    Create a library that is fully managed within VUnit. VUnit compiles files into it and manages the physical location. Users are not allowed to choose the location. There are no problems here with nvc.

2. `add_external_library(library_name, path)`
    Add a pre-compiled black-box library with arbitrary location. Since the location is arbitrary it might not be possible to use directly with nvc without work-arounds such as copying the folder. We could also forbid the path leaf name from differing from the library_name but then there is another worry. Since it is located in an arbitrary path that would have to be added to nvc library search paths with the `-L` flag we cannot control what other libraries there might be in the parent folder that accidentally also will get exposed.

**Example usage of VUnit library methods:**
```
prj.add_external_library("unisim", "/path/to/unisim/")
lib = prj.add_library("lib")
lib.add_source_files("*.vhd")
```

Even though my worries stated above are probably minor and rare cases they still lead me to prefer the interface of other simulators which lets you explicitly specify the mapping of library names to physical paths and not relying on implicit methods.

Just wanted to let you know my perspective...

nickg, 2015-05-09:
OK, so the proposed `--work=logical_name:path` syntax would cover the above usage adequately?

kraigher, 2015-05-09:
@nickg Yes is would solve the problem of the use of an arbitrary path in `add_external_library`. However the "problem" with the `-L` flag still remains. One would still have to add the entire parent folder of a library to the library search path when analyzing dependent files thus exposing every other library in that parent folder as a side-effect. There is no way to specifically say **exactly** what library names and paths should be used during analysis without changing location of directories.  Having a `library.cfg` file would solve that and also make the command line a bit shorter when there are many libraries and and thus many occurances of  `-L` flag arguments.

**Example libraries.cfg:**
```
library_name1 = path1
library_name2 = path2
```

kraigher, 2015-05-09:
@nickg An alternative to libraries.cfg would be to make the `-L` flag also support a `-L=library_name:path` syntax enabling explicit specification of all dependent libraries when analyzing a file.

nickg, 2015-05-09:
@kraigher That's probably easier to do in the short term. Rather than overloading `-L` how about adding a new option `--map=name:path` to do this?

kraigher, 2015-05-09:
@nickg the map flags sounds good to specify mapping of library dependencies. But wouldnt the work flag also have to change as we discussed to interact with this when specifying the library name to analyze into.

nickg, 2015-05-09:
@kraigher Yes I'll add both.

kraigher, 2015-05-09:
@nickg Also with the typical option of many tools to read command line flags with `-f` one could create the equivalent of libraries.cfg with this.

kraigher, 2015-05-09:
Read command line flags from file I mean.

nickg, 2015-05-09:
I've added these two as described. Seems to work OK in a quick test. Can you see if this works for your use case?

kraigher, 2015-05-09:
@nickg I am trying it out right now. Does not seem to work quite right. I will explore.
One first bug I found was this:

```
> nvc --work lib some_file.vhd
> ls -l
lib
some_file.vhd

> nvc --work lib:lib_path some_file.vhd
> ls -l
lib
some_file.vhd

> rm -rf lib
> nvc --work lib:lib_path some_file.vhd
> ls -l
lib_path
some_file.vhd 
```

nickg, 2015-05-09:
@kraigher That problem should be fixed now

kraigher, 2015-05-09:
@nickg I successfully used this feature now. Thank you! One last question: I guess the intention is to let the `--map` option take precedence over any library found in the library search paths?

nickg, 2015-05-09:
@kraigher Yes it ought to

closed 2015-05-09 by nickg

kraigher, 2015-05-09:
@nickg Great thank you!