nickg/nvc #163: Bug: Broken module on "wait until" statement.

kraigher, 2015-05-06:
```vhdl
package wait_until_pkg is
end package;

package body wait_until_pkg is
  procedure wait_until(signal sig : in boolean; val : boolean) is
  begin
    wait until sig = val; -- This does not work
  end procedure;

  procedure wait_until_works(signal sig : in boolean) is
  begin
    wait until sig = true; -- This works
  end procedure;
end package body;
```

```
> nvc -a wait_until.vhd 
AddrSpaceCast source must be a pointer
  %14 = addrspacecast i1 %10 to i8*
AddrSpaceCast source must be a pointer
  %22 = addrspacecast i1 %10 to i8*
Broken module found, verification continues.
AddrSpaceCast source must be a pointer
  %14 = addrspacecast i1 %10 to i8*
AddrSpaceCast source must be a pointer
  %22 = addrspacecast i1 %10 to i8*
Broken module found, verification continues.
Broken module found, verification continues.
AddrSpaceCast source must be a pointer
  %14 = addrspacecast i1 %10 to i8*
AddrSpaceCast source must be a pointer
  %22 = addrspacecast i1 %10 to i8*
Broken module found, verification continues.
Broken module found, verification continues.
Broken module found, verification continues.
** Fatal: LLVM verification failed 
```

kraigher, 2015-05-06:
This also works:

```vhdl
  procedure wait_until(signal sig : in boolean; val : boolean) is
    variable val_local : boolean := val;
  begin
    wait until sig = val_local; -- This works
  end procedure;
```

nickg, 2015-05-06:
Should be fixed now. Thanks for generating all these bug reports BTW!

closed 2015-05-06 by nickg

kraigher, 2015-05-07:
@nickg No problem. I am trying to add support for nvc to the [VUnit](https://github.com/LarsAsplund/vunit) VHDL unit testing framework which I am one of the authors of. The bugs I find are from trying to analyze the code. I have not got to running any of it yet but it is probably close now. We have quite complicated VHDL code due to working arround the limitations of VHDL to implement a unit testing tool with a reasonable public user interface. Some bugs are also found when I condense the other bugs I find and then I try to report them as well. 

Anyway thank you for your efforts! I think this is the most promising open source VHDL simulator. It will be really useful to the VHDL community if you keep at it.