nickg/nvc #164: Bug: LLVM broken module on procedure and function with the same name.

kraigher, 2015-05-06:
```vhdl
package same_name_broken_module is
end package;

package body same_name_broken_module is
  procedure same_name is
  begin
  end;

  function same_name return integer is
  begin
    return 0;
  end function;
end package body;
```

```
> nvc -a same_name_broken_module.vhd 
Function return type does not match operand type of return inst!
  ret i32 -2147483648
 i8*Broken module found, verification continues.
Function return type does not match operand type of return inst!
  ret i32 -2147483648
 i8*Broken module found, verification continues.
Broken module found, verification continues.
** Fatal: LLVM verification failed
```

nickg, 2015-05-08:
Interestingly I don't get this error: it's possibly dependent on the LLVM version. Which do you have? Also can you try running it with `--dump-llvm`? I get the following output:

```
; ModuleID = 'WORK.SAME_NAME_BROKEN_MODULE-body'

@module_name = private global [34 x i8] c"WORK.SAME_NAME_BROKEN_MODULE-body\00"
@_tmp_stack = external global i8*
@_tmp_alloc = external global i32

define void @WORK.SAME_NAME_BROKEN_MODULE-body_reset() {
vcode_block_0:
  ret void
}

; Function Attrs: nounwind
define i8* @WORK.SAME_NAME_BROKEN_MODULE.SAME_NAME(i8*) #0 {
vcode_block_0:
  ret i8* null
}

; Function Attrs: nounwind readonly
define i32 @"WORK.SAME_NAME_BROKEN_MODULE.SAME_NAME$I"() #1 {
vcode_block_0:
  ret i32 0
}

attributes #0 = { nounwind }
attributes #1 = { nounwind readonly }
```

kraigher, 2015-05-08:
The submitted report was based on nvc  built with LLVM 3.4:
I have rebuilt nvc using LLVM 3.3, 3.4 and 3.5 and get the same result:

```
# Build using LLVM 3.4
> nvc --version
nvc 0.1 (llvm 3.4; tcl 8.6)
> nvc -a same_name_broken_module.vhd  --dump-llvm
; ModuleID = 'WORK.SAME_NAME_BROKEN_MODULE-body'

@module_name = private global [34 x i8] c"WORK.SAME_NAME_BROKEN_MODULE-body\00"
@_tmp_stack = external global i8*
@_tmp_alloc = external global i32

define void @WORK.SAME_NAME_BROKEN_MODULE-body_reset() {
vcode_block_0:
  ret void
}

; Function Attrs: nounwind readonly
define i8* @"WORK.SAME_NAME_BROKEN_MODULE.SAME_NAME$I"(i32*, i8*) #0 {
vcode_block_0:
  ret i8* null

vcode_block_01:                                   ; No predecessors!
  %VAR = alloca i32
  store i32 -2147483648, i32* %VAR
  ret i32 -2147483648
}

attributes #0 = { nounwind readonly }
Function return type does not match operand type of return inst!
  ret i32 -2147483648
 i8*Broken module found, verification continues.
Function return type does not match operand type of return inst!
  ret i32 -2147483648
 i8*Broken module found, verification continues.
Broken module found, verification continues.
** Fatal: LLVM verification failed
```

```
# Build using LLVM 3.5
> nvc --version
nvc 0.1 (llvm 3.5.0; tcl 8.6)
> nvc -a same_name_broken_module.vhd  --dump-llvm
; ModuleID = 'WORK.SAME_NAME_BROKEN_MODULE-body'

@module_name = private global [34 x i8] c"WORK.SAME_NAME_BROKEN_MODULE-body\00"
@_tmp_stack = external global i8*
@_tmp_alloc = external global i32

define void @WORK.SAME_NAME_BROKEN_MODULE-body_reset() {
vcode_block_0:
  ret void
}

; Function Attrs: nounwind readonly
define i8* @"WORK.SAME_NAME_BROKEN_MODULE.SAME_NAME$I"(i32*, i8*) #0 {
vcode_block_0:
  ret i8* null

vcode_block_01:                                   ; No predecessors!
  %VAR = alloca i32
  store i32 -2147483648, i32* %VAR
  ret i32 -2147483648
}

attributes #0 = { nounwind readonly }
Function return type does not match operand type of return inst!
  ret i32 -2147483648
 i8*** Fatal: LLVM verification failed
```

```
Built using LLVM 3.3
> nvc --version
nvc 0.1 (llvm 3.3; tcl 8.6)
> nvc -a same_name_broken_module.vhd  --dump-llvm
; ModuleID = 'WORK.SAME_NAME_BROKEN_MODULE-body'

@module_name = private global [34 x i8] c"WORK.SAME_NAME_BROKEN_MODULE-body\00"
@_tmp_stack = external global i8*
@_tmp_alloc = external global i32

define void @WORK.SAME_NAME_BROKEN_MODULE-body_reset() {
vcode_block_0:
  ret void
}

; Function Attrs: nounwind readonly
define i8* @WORK.SAME_NAME_BROKEN_MODULE.SAME_NAME__I(i32*, i8*) #0 {
vcode_block_0:
  ret i8* null

vcode_block_01:                                   ; No predecessors!
  %VAR = alloca i32
  store i32 -2147483648, i32* %VAR
  ret i32 -2147483648
}

attributes #0 = { nounwind readonly }
Function return type does not match operand type of return inst!
  ret i32 -2147483648
 i8*Broken module found, verification continues.
Function return type does not match operand type of return inst!
  ret i32 -2147483648
 i8*Broken module found, verification continues.
Broken module found, verification continues.
** Fatal: LLVM verification failed
```

I ran it with the latest nvc 7176e5344b47148ef7aa73342d19779cfde0dc94.

nickg, 2015-05-08:
Is `same_name_broken_module.vhd` exactly the same code as posted in the first comment above? It seems to have an extra `VAR` variable.

kraigher, 2015-05-08:
@nick Ahh my bad, it should be:
```vhdl
package same_name_broken_module is
end package;

package body same_name_broken_module is
  procedure same_name (variable var : out integer) is
  begin
  end;

  impure function same_name return integer is
    variable var : integer;
  begin
    return var;
  end function;

end package body;
```

kraigher, 2015-05-08:
@nickg The code I originally posted also works for me. I must have made some mistake when creating the bug report.

nickg, 2015-05-08:
Fixed by the above commit. You'll need to do `make bootstrap install` as the name mangling format changed.

closed 2015-05-08 by nickg

kraigher, 2015-05-08:
@nickg I always do `make clean` when I update. Should be enough?

nickg, 2015-05-08:
Yes. `make bootstrap` just cleans and rebuilds the libraries.