nickg/nvc #168: Feature request: Emit VHDL call stack trace.

kraigher, 2015-05-07:
When debugging VHDL code it is very useful to get the call stack trace on for instance a failing assert.
This is found in most commercial simulators. Without a call stack trace it can be very hard to determine where the error occured if re-usable sub-programs like `check` or `check_equal` are used. 

Consider the following code:
```vhdl
entity test is
end entity;

architecture a of test is
  procedure check(value : boolean) is
  begin
    assert value;
  end procedure;
begin

  main : process
    procedure do_some_other_stuff is
    begin
      check(false);
    end procedure;

    procedure do_stuff is
    begin
      check(true);
      do_some_other_stuff;
      check(true);
    end procedure;
  begin
    check(true);
    do_stuff;
  end process;
end architecture;
```

```
> nvc -a test.vhd && nvc -e test && nvc -r test
** Fatal: 0ms+0: Assertion Error: Assertion violation.
	Process :test:main
	File test.vhd, Line 7 
```

The error message just tells me an assert has failed in my re-usable check function which is not very helpful if it is used everywhere. It would be great to get a full stack trace such as:
```
** Fatal: 0ms+0: Assertion Error: Assertion violation.
	Process :test:main
	File test.vhd, Line 7
        ... called from test.vhd:14 do_some_other_stuff
        ... called from test.vhd:20 do_stuff
        ... called from test.vhd:25 :test:main
```

Other simulators expose this as a TCL command that you can execute after simulation has stopped on an error or any other reason. Having it enabled by a command line flag would also work.

A simple implementation of this would be to store a static table of line numbers and strings to all possible call locations and pushing/poping indexes to this table when calling and returning from subprograms. For people who do not want to pay for the runtime overhead maybe a flag has to be added such that it can be disabled thus also disabling the possibility to have a stack trace.

nickg, 2015-05-08:
Yes it's a good idea. I'd already been thinking about doing this with LLVM's Debug Info.

nickg, 2021-05-09:
I finally got round to adding this, but it only works on Linux for now and requires `libdw-dev` which is currently an optional dependency.

```
$ nvc -a test.vhd -e test -r
/usr/bin/ld --eh-frame-hdr -shared -o /tmp/work/_WORK.TEST.elab.so /tmp/work/_WORK.TEST.elab.o
** Note: in process :test:main
	File test.vhd, Line 25
        do_stuff;
        ^^^^^^^^^
** Note: in subprogram DO_STUFF []
	File test.vhd, Line 20
          do_some_other_stuff;
          ^^^^^^^^^^^^^^^^^^^^
** Note: in subprogram DO_SOME_OTHER_STUFF []
	File test.vhd, Line 14
          check(false);
          ^^^^^^^^^^^^^
** Fatal: 0ms+0: Assertion Error: Assertion violation.
	Process :test:main
	File test.vhd, Line 7
        assert value;
        ^^^^^^^^^^^^^
```

closed 2021-05-09 by nickg