nickg/nvc #170: ambiguous use of operator "=" in ieee.std_logic_1164 and ieee.std_logic_unsigned.

fmhess, 2015-05-07:
The following program fails to analyze, due to ambiguity on the "=" operator when using both ieee.std_logic_1164 and ieee.std_logic_unsigned.  Using the suggested "--relax=prefer-explicit" option does make the error go away, but it seems like built-in libraries should just work.

```
library IEEE;
use IEEE.std_logic_1164.all;
use ieee.std_logic_unsigned.all;

entity testbench is
end testbench;
     
architecture behav of testbench is
  signal data          : STD_LOGIC_VECTOR (7 DOWNTO 0);
begin
	process is
	begin
	if (data = "00000001") then
	end if;
	end process;
end behav;
```

The error output is:
```
$ nvc -a test.vhd
** Error: ambiguous use of operator "="
              "=" [STD_LOGIC_VECTOR, STD_LOGIC_VECTOR return BOOLEAN] (implicit)
              "=" [STD_LOGIC_VECTOR, STD_LOGIC_VECTOR return BOOLEAN]
          You can use the --relax=prefer-explicit option to hide the implicit
          operator
        File test.vhd, Line 13
     if (data = "00000001") then
         ^^^^^^^^^^^^^^^^^
```

nickg, 2015-05-07:
The Synopsys libraries are a legacy artefact which NVC provides for convenience and are not part of the IEEE standard. The above error is correct according to the LRM. Because of this and other problems the IEEE standardised the numeric_std library which should be used instead.

closed 2015-05-07 by nickg

fmhess, 2015-05-07:
Understood.  I would use numeric_std, but I'm dealing with legacy code written (and maintained) by someone else.  I'll just use the relax option.  nvc has been a tremendously useful tool for me btw, thanks!

nickg, 2015-05-07:
Great. The `--relax` option is for cases just like this where the standard says something is incorrect but other tools allow it, etc.