nickg/nvc #171: Feature request: Set top level generics from command line during elaboration.

kraigher, 2015-05-08:
Many verification environments rely on being able to set test bench generics from a command line argument during elaboration. Both to re-use the same test bench using multiple generic configurations of the DUT (Device under test) but also for having multiple test cases in the same test bench running in different simulations but re-using the infrastructure of the test bench. 

The [VUnit](https://github.com/LarsAsplund/vunit) VHDL test framework relies on being able to set `string` generics from command line to pass information from the driving verification script to the VHDL test runner.

Adding a `-g` flag to set top level generics during elaboration would be good enough:
```
nvc -e lib.test_bench -g runner_cfg=... -g data_width=11
```
```vhdl
entity test_bench is
  generic (
    runner_cfg : runner_cfg_t; -- runner_cfg_t is alias/subtype of string
    data_width : positive);
end entity;
```

nickg, 2015-05-13:
Setting generics at elaboration time should be straightforward: I can add the `-gX=Y` syntax as above. It should also be possible to set generics at runtime, provided their value wasn't needed at elaboration time (e.g. for a generate generate statement) and they were marked as such at elaboration (e.g. `-gX=?`).

kraigher, 2015-05-13:
@nickg Setting generics at runtime would be nice to save having to re-elaborate the entire design hierarchy for a testbench with say 10 test cases which are selected by a generic.
It would be hard for outside driving scripts such as VUnit to know if it has to re-elaborate using `-gX=Y` or can just re-run with `-gX=Y`. GHDL has an --elab-run flag which does both elaborate and run. With such a flag the simulator could itself know if it can re-use elaboration results.

nickg, 2015-05-31:
I've added support for this at elaboration time. Adding runtime generic overrides requires quite a lot more work so I'll leave that for the future.

closed 2015-05-31 by nickg