nickg/nvc #172: Feature request: Add VHDL-2008 std.textio.tee to support latest OSVVM (2015.03)

kraigher, 2015-05-08:
The latest OSVVM (2015.03) contains a TranscriptPkg.vhd which uses the new TEE procedure added to std.textio in VHDL 2008:

```
-- writes to file F *and* to standard output
procedure TEE (file F: TEXT; L: inout LINE);
```

peteut, 2016-04-02:
This can be resolved by using the `ieee_proposed.standard_textio_additions` package.

kraigher, 2017-07-30:
A question regarding this, would you mind accepting a pull request adding the `TEE` function to `std.textio`?
Do you care that it will be available also for other standards than 2008? Is the goal of NVC just to be good at the latest VHDL standard without caring much with limiting the availability of new features when running with old standards? To be this would be a reasonable trade-off but I know GHDL tries to be very exact with the standard by limiting features.

nickg, 2017-07-30:
I think it's fine as it won't break any existing code.

closed 2017-07-31 by kraigher