nickg/nvc #173: Bug: Work in fully qualified name not treated as reference to current library.

kraigher, 2015-05-08:
```vhdl
package other_pkg is
  type type_t is (value1, value2);
  function fun return integer;
  procedure proc;
end package;

package body other_pkg is
  function fun return integer is
  begin
    return 0;
  end function;

  procedure proc is
  begin
  end procedure;
end package body;

use work.other_pkg.all;

package pkg is
end package;

package body pkg is
  alias pkg_t is work.other_pkg.type_t;
  procedure proc is
    variable var : integer;
  begin
    work.other_pkg.proc;
    var := work.other_pkg.fun;
  end procedure;
end package body;
```

```
> nvc --work lib -a work_is_special.vhd
** Error: no visible declaration for WORK.OTHER_PKG.TYPE_T
	File work_is_special.vhd, Line 24
      alias pkg_t is work.other_pkg.type_t;
                     ^^^^^^^^^^^^^^^^^^^^^
** Error: undefined procedure WORK.OTHER_PKG.PROC
	File work_is_special.vhd, Line 28
        work.other_pkg.proc;
        ^^^^^^^^^^^^^^^^^^^^
** Error: no visible declaration for WORK.OTHER_PKG.FUN
	File work_is_special.vhd, Line 29
        var := work.other_pkg.fun;
               ^^^^^^^^^^^^^^^^^^
```

Calling the library `work` makes it work:
```
> nvc --work work -a work_is_special.vhd
```

This makes me believe that the library name `work` is not treated specially in this context as it should. As you probably know `work` is not a valid library logical name but instead an automatic reference to the  library into which the current design unit is being compiled. Even allowing the user to call his library work with `--work work` is a bit of a anti-pattern which unfortunately other simulators also allow. See this post for more info http://www.sigasi.com/content/work-not-vhdl-library .

closed 2015-05-09 by nickg

kraigher, 2015-05-09:
@nickg Just out of interest, what is your opinion on allowing work as a library logical name? I know you have selected it as the default library logical name in your simulator as have for instance modelsim and that it might be a common convention. It is kind of an misuse of the language though which can cause problems that people do not understand since they often have the wrong idea about work and think it is just a pre-defined default library. In VUnit we have chosen to forbid having work as a library name and telling users why in an error message. Xilinx use `xil_defaultlib` as the default library name in their tools btw.

nickg, 2015-05-09:
It's probably an unfortunate default as it hides issues like this. It comes from me mostly having used Modelsim until I started working on NVC where using `vlib work` is ubiquitous. And as someone doing FPGA design rather than tool implementation I wasn't really aware of the details when I started working on NVC (and the library code is ~4 years old). 

That said, the LRM doesn't explicitly _disallow_ naming a library `work`: all it has is

> Library logical name WORK denotes the current working library during a given analysis.

I guess the best solution would be to allow overriding the default in a global or project INI file.