nickg/nvc #175: Bug: Elaboration problem with work library reference.

kraigher, 2015-05-08:
```vhdl
package pkg2 is
  constant const2 : integer := 1;
end package;

package body pkg2 is
end package body;
```

```vhdl
use work.pkg2.all;

package pkg is
  constant const : integer := const2 + 1;
end package;

package body pkg is
end package body;
```

```vhdl
library lib2;
use lib2.pkg.all;

entity ent is
end entity;

architecture a of ent is
begin
  main : process
  begin
    assert const = 2;
    wait;
  end process;
end architecture;
```

```
# test.sh
set -ex
rm -rf lib2
rm -rf lib
nvc -L . --work=lib2 -a pkg2.vhd
nvc -L . --work=lib2 -a pkg.vhd
nvc -L . --work=lib -a entity.vhd
nvc -L . --work=lib -e ent
```

```
> sh test.sh
sh test.sh 
+ rm -rf lib2
+ rm -rf lib
+ nvc -L . --work=lib2 -a pkg2.vhd
+ nvc -L . --work=lib2 -a pkg.vhd
+ nvc -L . --work=lib -a entity.vhd
+ nvc -L . --work=lib -e ent
** Fatal: cannot find unit WORK.PKG2
```

It seems the work reference is not translated to the correct library name.

nickg, 2015-05-09:
This one should be OK now.

closed 2015-05-09 by nickg

kraigher, 2015-05-09:
@nickg It still fails for me when running the same `test.sh`.

nickg, 2015-05-09:
Oops, yes, it fails in a slightly different place which isn't caught by the unit test. Should be fixed now.

kraigher, 2015-05-09:
@nickg Running `test.sh` works now. Thank you!