nickg/nvc #179: Bug: Design unit older than source file check causes problems.

kraigher, 2015-05-09:
```
** Fatal: design unit WORK.TB_FOO is older than its source file
          tb_foo.vhd
          and must be reanalysed
```
Maybe it is a matter if interpretation if this is a bug or not, but having such a check can cause problems.
In VUnit we do not re-compile files if the sha1 hash of the contents has not changed rather than relying on the timestamp. When doing git checkout or git rebase it changes the timestamp of all files but the content might not change and they do not have to be recompiled. Our dependency scanner will not re-compile such files and cause nvc to complain.

Also philosophically having a check of the timestamp of the source file and halting on error by the simulator is malplaced and too interfering in my opinion. It should be the choice of the user if he/she wants to recompile the file or not. Maybe the content has changed but the user does not want to re-analyze it anyway.

nickg, 2015-05-09:
While the above are all good points I actually find this feature quite useful when using `nvc` interactively (i.e. not through a script or makefile). Then it's helpful to be reminded when a unit is out of date. How about if there was an `--ignore-time` option that disabled this check?

kraigher, 2015-05-09:
@nickg Yes adding a flag to remove the failure would solve my problem with it.

kraigher, 2015-05-09:
@nickg You probably already thought of it but mentioning the flag in the error message would be nice for users.

nickg, 2015-05-09:
Added that. Now it prints:

```
** Fatal: design unit WORK.WAIT1 is older than its source file ../test/regress/wait1.vhd
          and must be reanalysed
          (You can use the --ignore-time option to skip this check)
```

closed 2015-05-09 by nickg