nickg/nvc #181: Bug: String with null range cause runtime error.

kraigher, 2015-05-09:
I am not sure if this violates the LRM or if it is a bug. Modelsim allows it but prints a warning. 

```vhdl
entity ent is
end entity;

architecture a of ent is
  procedure proc is
    variable str : string(1 to 0);
    variable bitv : bit_vector(1 to 0); -- Works
  begin
  end procedure;
begin
  main : process
  begin
    proc;
    wait;
  end process;
end architecture;
```

```
> nvc -a string_null_range.vhd && nvc -e ent && nvc -r ent
/usr/lib/llvm-3.5/bin/opt -O2 -o /home/kraigher/repo/nvc_bugs/string_null_range/work/_WORK.ENT.final.bc /home/kraigher/repo/nvc_bugs/string_null_range/work/_WORK.ENT.elab.bc
** Fatal: index 0 violates constraint bounds 1 to 2147483647
	File string_null_range.vhd, Line 6
```

nickg, 2015-05-09:
I think this should be OK according to the LRM. In section 5.2.1 or the 2008 version it says

> A range constraint is compatible with a subtype if each bound of the range belongs to the subtype or if the range constraint defines a null range. Otherwise, the range constraint is not compatible with the subtype.

The index constraint for `STRING` has subtype `POSITIVE` which is why 0 causes a problem (for `BIT_VECTOR` it's `NATURAL`). However `1 to 0` is a null range so should be acceptable according to the above.

What warning does Modelsim give?

kraigher, 2015-05-09:
```
** Warning: [3] string_null_range.vhd(6): (vcom-1246) Range 1 to 0 is null.
** Warning: [3] string_null_range.vhd(7): (vcom-1246) Range 1 to 0 is null.
```

closed 2015-05-12 by nickg