nickg/nvc #184: Bug: String comparison in if-generate statement cannot be folded to integer constant.

kraigher, 2015-05-10:
```vhdl
entity ent is
  generic (
    config : string := "config0");
end entity;

architecture a of ent is
  signal sig : integer;
begin
  gen_cfg1 : if config = "config1" generate
    sig <= 0;
  end generate;
end architecture;
```

```
> nvc -a fold_to_integer_constant.vhd && nvc -e ent
** Fatal: expression cannot be folded to an integer constant
	File fold_to_integer_constant.vhd, Line 9
```

kraigher, 2015-05-10:
This also fails:

```vhdl
entity ent is
  generic (
    config : bit_vector := "0");
end entity;

architecture a of ent is
  signal sig : integer;
begin
  gen_cfg1 : if config = "1" generate
    sig <= 0;
  end generate;
end architecture;
```

nickg, 2015-05-13:
I've put in something temporary to fix this. I'm currently in the process of rewriting the constant evaluation code to be much more generic, and based off the same vcode module that does code generation.

closed 2015-05-13 by nickg