nickg/nvc #189: Question: stop nvc simulation after last event

uffejakobsen, 2015-05-12:
Hi,
Apologies if this question it just too plain stupid - but my vhdl-foo is a bit rusty
I have a simple top level test bench - and I want to stop simulation after x clock "iterations"
From other vhdl simulation tools I seem to remember that they stop when no events are generated.
This seems not to be the case with nvc

PS: I know about the --stop-time option - but I would like the test benches to be self contained - and hence not use external parameters.
PPS: I would also like to avoid using assert to stop the test bench

Question: will nvc simulation stop after last event ?

Thanks in advance :-)

nickg, 2015-05-12:
The simulation will stop when there are no more events. You can try `nvc -r --trace` to get some clue of which events are being generated.

uffejakobsen, 2015-05-12:
Thanks for your quick reply.

Running nvc simulation with --trace option shows that no more events are being generated after the specified time (at least no more output to stdout).
But execution of nvc itself does not stop
nvc does not exit back to shell prompt - it still consumes one full cpu "forever"

nickg, 2015-05-12:
What's the final output of the `--trace` command? Is it possible that you have a process which is not suspending?

uffejakobsen, 2015-05-12:
Sorry my vhdl is a bit rusty - I'm I've overlooked something.

Last stdout output is:
TRACE 7960ns+1: _vec_load :ef3_led_tb:clk where=0x7fff8fde1440 low=0 high=0 last=1
TRACE 8us+0: begin cycle
8us	process	 :ef3_led_tb:line_29
TRACE 8us+0: run process :ef3_led_tb:line_29

nickg, 2015-05-12:
It looks like the process around line 29 of `ef3_led_tb` is stuck in an infinite loop. If you paste it here I might be able to help...

uffejakobsen, 2015-05-12:
The is the process from my tb at line 29: - i have no idea how to break out of the loop - my only idea was to avoid generating more events after 200 cycle iterations.... :-S

  process
  begin
    if (clk_cnt < 200) then
      clk_cnt <= clk_cnt + 1;
      clk <= not clk;
      wait for 40ns;
    end if;
  end process;

nickg, 2015-05-12:
So if `clk_cnt >= 200` then that process just loops forever. If you reach the end of the process statements execution starts again from the top of the process. Something like this should fix it:

```vhdl
process
begin
    if (clk_cnt < 200) then
        clk_cnt <= clk_cnt + 1;
        clk <= not clk;
        wait for 40ns;
    end if;
    wait;
end process;
```

nickg, 2015-05-12:
Oops, sorry, that should be:

```vhdl
process
begin
    if (clk_cnt < 200) then
        clk_cnt <= clk_cnt + 1;
        clk <= not clk;
        wait for 40ns;
    else
        wait;
    end if;
end process;
```

uffejakobsen, 2015-05-12:
Ah - I missed that detail - embarrasing - thanks  :-) :+1: 
Sorry it has been too long ago I've used vhdl - I need to go and refresh my vhdl skills elsewhere
Any suggestions for eg. a IRC channel or something that will take these kind of stupid questions ?

Thanks for your help and sorry for wasting your time :-)

nickg, 2015-05-12:
You could try the VHDL tag on stackoverflow: that's usually quite active

closed 2015-05-12 by nickg

uffejakobsen, 2015-05-12:
ok thanks :-)

nickg, 2015-05-12:
FWIW I added something to print out the currently active process when you hit ctrl-c:

```
^C** Fatal: interrupted in process :ram1:stim at 1350ns+0
        File ../test/regress/ram1.vhd, Line 71
```

Should make this sort of thing easier to debug.

uffejakobsen, 2015-05-12:
Cool thanks - will do a rebuild now - it isquite useful for someone like me :-)

kraigher, 2015-05-13:
Some simulators support setting the max number of delta cycle iterations that are allowed without advancing time and the simulator will error out or give a warning if it is exceeded. In ModelSim the default iteration limit is 5000 and it can be changed by setting the `IterationLimit` TCL variable. This will catch processes like this:
```vhdl
p :  process
begin
end process; 
```

But not these:
```vhdl
p :  process
begin
  loop
  end loop;
end process; 
```

kraigher, 2015-05-13:
@uffejakobsen A better way to achieve what you want is to use a `signal : finished : boolean := false` in your test bench and having the clock process look like this:
```vhdl
clk_gen : process
begin
  while not finished loop
    clk <= not clk;
    wait for 1 ns;
  end loop;
  wait;
end process;
```
The finished signal would be set to `true` by your main process or test sequencer after all tests are done. Thus when adding new tests you will not have to change the hardcoded clock cycle count all the time which is kind of equivalent to setting a hardcoded runtime except that it is done in the VHDL.

An even better way would be to use the `std.env.stop` procedure of VHDL-2008 which NVC supports (as well as ModelSim, Aldec and GHDL). There is really no good reason to not use VHDL-2008 for simulation. Then your main sequencer can just do stop after it is done:
```
use std.env.stop;
-- ...
main : process
begin
  -- ...
  stop(0);
end process;
```

You can more about `stop` here: https://www.doulos.com/knowhow/vhdl_designers_guide/vhdl_2008/vhdl_200x_small/

nickg, 2015-05-13:
@kraigher Reaching the end of a process causes it to restart immediately rather than on the next delta cycle. See the 2008 LRM section 11.3:

> After the last statement in the sequence of statements of a process statement is executed, execution will immediately continue with the first statement in the sequence of statements.

NVC has a `--stop-delta` option to stop after a certain number of delta cycles, which defaults to 1000, and would catch errors like:

```vhdl
    process (x) is
    begin
        x <= x + 1;
    end process;
```

```
** Fatal: Iteration limit of 1000 delta cycles reached. The following processes
          are active:
            :test:line_8                   test.vhd line 8
          You can increase this limit with --stop-delta
```

kraigher, 2015-05-13:
@nickg Aha. I did not know that. Then no complaints can be made against nvc.