nickg/nvc #192: Feature request: Support for embedding NVC in other programs

forrestv, 2015-05-29:
I'd like to be able to run VHDL code from an external C/C++ program (not arbitrary code, but just pre-analyzed/elaborated/generated code) because I want the code's foreign function calls to be within the context of my program and not a subprocess.

Since I doubt you want to expose/document a full API, the best solution to this that I can see is exposing a single function call - `nvc_main(argc, argv)` - in a shared library or something along those lines. And, IMO, the best way to do that would be to convert the entire project to generating the shared library, plus a tiny stub `nvc` binary that just calls `nvc_main`. This is slightly complicated by `nvc` registering signal handlers by default currently, but perhaps a third argument could provide a way to disable that.

Does that sound reasonable? If so, I can implement it, but I am curious to hear your thoughts on the matter.

nickg, 2015-05-29:
TBH I'm not very keen on this. Exposing just the `main` function seems quite ugly and doesn't seem enough to justify switching the build system to use `libtool`, etc. (and `libtool` is *really* nasty). Adding an API is something I might consider doing in the future, but not for the time being. There's also the issue of the GPL: to provide a useful shared library most users will want LGPL or they'll end up with unexpected licensing issues.

FWIW I think you can achieve almost exactly what you want by compiling `nvc` with `-fpie` in `CFLAGS` (and maybe `-pie` in `LDFLAGS`) and then using `dlopen` in your program and getting the address of the `main` function with `dlsym`. I'd be happy to accept patches to add those flags via the configure script (`--enable-relocatable` or something like that) and some way of disabling the signal handlers with e.g. a `configure` flag or an environment variable (note LLVM JIT installs its own handlers so you'd have to elaborate the VHDL with `--native` to avoid this).

nickg, 2018-02-06:
If you just want to call the `main` function, use the technique described above, otherwise the official extension mechanism is VHPI. There are no plans to add a full API at  the moment.

closed 2018-02-06 by nickg