nickg/nvc #193: ** Fatal: cannot heap allocate

kraigher, 2015-05-31:
**bug.vhd:**

```vhdl
entity ent is
end entity;

architecture a of ent is

  procedure other_proc(arg : boolean) is
  begin
    wait for 0 ns;
    report boolean'image(arg);
  end procedure;

  procedure proc(variable arg : integer) is
  begin
    other_proc(arg = 0);
  end procedure;
begin
  main : process
  begin
    wait;
  end process;
end architecture;
```

**command line output:**
```console
nvc -a bug.vhd && nvc -e ent && nvc -r ent

Name       :ent:proc$vI
Kind       procedure
Context    WORK.ENT.elab
Blocks     1
Registers  3
Types      3
Variables  0
Parameters 1
  r0    ARG                             // -2^31..2^31-1 
Begin
   0: r1 := const 0                     // -2^31..2^31-1 => 0 
      r2 := cmp r0 == r1
      // Elided bounds check for r2 

** Fatal: cannot heap allocate r2

-------- STACK TRACE --------
nvc(show_stacktrace+0x16) [0x415016]
nvc(fatal_trace+0x9e) [0x4150ee]
nvc(vcode_heap_allocate+0x123) [0x464bd3]
nvc() [0x45e3fc]
nvc() [0x45f5c9]
nvc(lower_unit+0x118) [0x460318]
nvc(main+0xe37) [0x411ac7]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf5) [0x7f2234c2fec5]
nvc() [0x412257]
-----------------------------
```

nickg, 2015-05-31:
Was broken by the fix to #185

closed 2015-05-31 by nickg

kraigher, 2015-05-31:
Found another similar problem even after this fix. Will investigate and make a new issue.

kraigher, 2015-05-31:
This also fails:
```vhdl
package pkg is
  type prot_t is protected
    procedure proc;
  end protected prot_t;

  procedure proc(variable prot : inout prot_t);
end package;

package body pkg is
  type prot_t is protected body
    procedure proc is
    begin
    end procedure;
  end protected body prot_t;

  procedure proc(variable prot : inout prot_t) is

    procedure nested is
    begin
      prot.proc;
    end procedure;
  begin
  end procedure;
end package body;
```

```console
 nvc --std=2002 -a bug.vhd

Name       WORK.PKG.PROC$vuWORK.PKG.PROT_T;__NESTEDv
Kind       procedure
Context    WORK.PKG.PROC$vuWORK.PKG.PROT_T;
Blocks     1
Registers  1
Types      0
Variables  0
Parameters 0
Begin
   0: r0 := param upref 1, r0           // @<WORK.PKG.PROT_T{}> => WORK.PKG.PROT_T{} 

** Fatal: cannot heap allocate r0

-------- STACK TRACE --------
nvc(show_stacktrace+0x16) [0x415016]
nvc(fatal_trace+0x9e) [0x4150ee]
nvc(vcode_heap_allocate+0x123) [0x464bd3]
nvc() [0x45e3fc]
nvc() [0x45f5c9]
nvc() [0x45f59c]
nvc(lower_unit+0xb6) [0x4602b6]
nvc(main+0x1314) [0x411fa4]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf5) [0x7f36d0accec5]
nvc() [0x412257]
-----------------------------
```

nickg, 2015-05-31:
The last one should be fixed by a232129

kraigher, 2015-05-31:
There is more:
```vhdl
package pkg is
  type prot_t is protected
    procedure proc;
  end protected prot_t;

  procedure proc(variable prot : inout prot_t; signal sig : in string);
end package;

package body pkg is
  type prot_t is protected body
    procedure proc is
    begin
    end procedure;
  end protected body prot_t;

  procedure proc(variable prot : inout prot_t;
                 signal sig : in string) is

    procedure nested(sig : string) is
    begin
      prot.proc;
    end procedure;
  begin
    nested(sig);
  end procedure;
end package body;
```

```console
nvc --std=2002 -a bug.vhd

Name       WORK.PKG.PROC$vuWORK.PKG.PROT_T;sS
Kind       procedure
Context    WORK.PKG-body
Blocks     1
Registers  11
Types      11
  WORK.PKG.PROT_T                       {}
Variables  0
Parameters 2
  r0    PROT                            // @<WORK.PKG.PROT_T{}> => WORK.PKG.PROT_T{} 
  r1    SIG                             // [*] : $<0..183> 
Begin
   0: r2 := unwrap r1                   // $<0..183> 
      r3 := uarray len r1 dim 0         // # 
      r4 := vec load r2 length r3       // @<0..183> => 0..183 
      r5 := uarray left r1 dim 0        // # 
      r6 := cast r5                     // -2^31..2^31-1 => 1..2^31-1 
      r7 := uarray right r1 dim 0       // # 
      r8 := cast r7                     // -2^31..2^31-1 => 1..2^31-1 
      r9 := uarray dir r1 dim 0         // 0..1 
      r10 := wrap r4 [r6 r8 r9]         // [*] : 0..183 

** Fatal: cannot heap allocate r4

-------- STACK TRACE --------
nvc(show_stacktrace+0x16) [0x415016]
nvc(fatal_trace+0x9e) [0x4150ee]
nvc(vcode_heap_allocate+0x123) [0x464bd3]
nvc() [0x45e3fc]
nvc() [0x45f5c9]
nvc(lower_unit+0xb6) [0x4602b6]
nvc(main+0x1314) [0x411fa4]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf5) [0x7f86ce378ec5]
nvc() [0x412257]
-----------------------------
```

nickg, 2015-05-31:
OK, hopefully that should be all of them.