nickg/nvc #196: Procedure argument corrupted.

kraigher, 2015-05-31:
This took a while to narrow down. Seems the arguments pushed on the stack cause memory corruption of another parameter. `var.field` should be zero also at second print but it is the bytes of `abcd`.

```vhdl
package pkg is
  procedure fun(s : string);
end package;

package body pkg is
  procedure fun(s : string) is
  begin
  end procedure;
end package body;

use work.pkg.all;

package pkg2 is
  type prot_t is protected
    procedure proc(s : string);
  end protected;
end package;

package body pkg2 is
  type prot_t is protected body
  procedure proc(s : string) is
  begin
    fun(s & LF);
  end procedure;
  end protected body;
end package body;

use work.pkg2.all;

entity ent is
end entity;

architecture tb of ent is
  type rec_t is record
    field : natural;
  end record;

  procedure print(r : rec_t) is
  begin
    report integer'image(r.field);
  end procedure;

  shared variable prot : prot_t;

  function fun return string is
  begin
    return "";
  end function;

  procedure proc is
  begin
    prot.proc("------------------------------------------------------------");
    wait for 0 ns;
    prot.proc("------------------------------------------------------------");
  end;

  procedure proc2 (
    constant var : in rec_t := (others => 0)) is
  begin
    wait for 0 ns;
    print(var);
    assert var.field = 0;
    prot.proc("--------------------------------abcd");  -- abcd corrupts var
    wait for 0 ns;
    print(var);

    -- var.field was overwritten by "abcd"
    report integer'image((((character'pos('d')*256 + character'pos('c'))*256 + character'pos('b'))*256 + character'pos('a')));
    assert var.field = ((((character'pos('d')*256) + character'pos('c'))*256 + character'pos('b'))*256 + character'pos('a'));
  end procedure;
begin
  main : process
  begin
    proc;
    proc2;
    wait;
  end process;
end architecture;
```

```console
nvc --std=2002 -a bug.vhd && nvc --std=2002 -e ent && nvc --std=2002 -r ent
/usr/lib/llvm-3.5/bin/llvm-link /home/kraigher/repo/nvc_bugs/corrupt/work/_WORK.ENT.elab.bc /home/kraigher/repo/nvc_bugs/corrupt/work/_WORK.PKG-body.bc /home/kraigher/repo/nvc_bugs/corrupt/work/_WORK.PKG2-body.bc -o /tmp/WORK.ENTRjavPn
/usr/lib/llvm-3.5/bin/opt -O2 -o /home/kraigher/repo/nvc_bugs/corrupt/work/_WORK.ENT.final.bc /tmp/WORK.ENTRjavPn
** Note: 0ms+2: Report Note: 0
	Process :ent:main
	File bug.vhd, Line 40
** Note: 0ms+3: Report Note: 1684234849
	Process :ent:main
	File bug.vhd, Line 40
** Note: 0ms+3: Report Note: 1684234849
	Process :ent:main
	File bug.vhd, Line 68
```

kraigher, 2015-06-01:
Here is a more minimal reproducer:
```vhdl
package pkg is
  procedure proc(s : string);
end package;

package body pkg is
  procedure proc(s : string) is
  begin
  end procedure;
end package body;

use work.pkg.all;
entity ent is
end entity;

architecture tb of ent is
  type rec_t is record
    field : natural;
  end record;

  procedure proc is
  begin
    proc("" & LF);
    wait for 0 ns;
    proc("------------------------------------------------------------" & LF);
  end;

  procedure proc2 (
    constant var : in rec_t := (others => 0)) is
  begin
    wait for 0 ns;
    report integer'image(var.field);
    assert var = (others => 0);
    proc("--------------------------------abcd" & LF);  -- abcd corrupts var

    -- var.field was overwritten by "abcd"
    report integer'image(var.field) & " = " & integer'image((((character'pos('d')*256 + character'pos('c'))*256 + character'pos('b'))*256 + character'pos('a')));
    assert var = (others => 0); -- Should still be 0
  end procedure;
begin
  main : process
  begin
    proc;
    proc2;
    wait;
  end process;
end architecture;
```

nickg, 2015-06-01:
Thanks for generating the test case!

closed 2015-06-01 by nickg

kraigher, 2015-06-02:
@nickg I was just wondering how stacks are implemented in NVC? I guess each suspended process needs a different stack since a wait statement can occur at any call depth. If each process has its own stack how do you allocate them and ensure that they do not overflow? In GHDL they have a fixed stack size for each process that you have to set using the `--stack-size` and `--stack-max-size` flags.

nickg, 2015-06-02:
@kraigher All processes in NVC share the same call stack: procedures and processes are implemented as state machines generated from the wait statements within them so a process is just a simple function call without having to allocate or switch call stacks (see `cgen_jump_table`). When you execute a wait statement in a procedure it simply returns out of all the nested procedures/processes to the kernel. This makes processes very cheap to create/execute.

The "stack" referred to here is actually a separate area used for allocating temporary objects. Normally this is shared between all processes but in the event that a process suspends inside a procedure while objects are still live in this we need to make a temporary copy. The vast majority of processes never suspend inside a procedure call so don't incur this overhead.

kraigher, 2015-06-02:
@nickg Does this mean that there are degenerate cases where excessive copying can occur for instance if a procedure allocates a big data structure on the stack and suspends? But at least access types are allocated on the heap with `new` right? Maybe it is true that processes in the synthesizeable code do not suspend in a procedure call it is not true for verification code in many cases. What about concurrent procedure calls? Are they converted into a process?

Consider the following not uncommon example, would it be a degenerate case:
```vhdl
bad_process : process
-- Data vector is large
  procedure write_data(constant data_vector : integer_vector) is
  begin
    for i in data_vector'range loop 
      valid <= '1';
      data <= to_unsigned(data_vector(i), data'length);
      wait until (valid and ready) = '1' and rising_edge(clk);
      valid <= '0';
      random_jitter;
    end loop;
  end procedure;

  procedure write_random_data(constant num : integer) is
    variable data_vector : integer_vector(0 to num-1);
  begin
    for i in 0 to data_vector'range loop
      data_vector(i) := rnd.RandInt(0, 2**data'length-1);
    end loop;
    write_data(data_vector);
  end procedure;
begin
  write_random_data(2**20);
  wait;
end process;
```

nickg, 2015-06-02:
@kraigher All concurrent statements are rewritten as processes: the LRM specifies the transformation.

NVC never copies the temporary stack it: just makes the active one private to that process and allocates a new one for future shared use. But in your example above that doesn't happen: variables in a procedure are part of its "state" so are placed in a heap allocated structure. (If `write_random_data` could never suspend it would be implemented as a regular function with `data_vector` on the call stack.) The "temporary stack" is really just for truly temporary data such as function results. E.g. if you did something like this:

```vhdl
write_data(some_function_returning_data(...));
```

Then the result of that function call would be stored on the temporary stack and needs to be preserved when `write_data` suspends.

kraigher, 2015-06-02:
@nickg I see. Thank you for your quick response.