nickg/nvc #198: Feature request: Improve performance by avoiding re-linking.

kraigher, 2015-06-02:
Consider the situation where a library of many VHDL packages containing verification code is used by every test bench and that there are hundreds of such test benches. Most test benches and their corresponding device under test are very small. It seems NVC will spend a lot of time re-linking everything including the re-usable verification library for every elaboration/run of the individual test benches making the regression suite take a very long time. When the run-time of each test bench is very small the link-time will dominate hugely. 

I have a regression suite where running the tests using ModelSim and GHDL-LLVM takes around 0.2 s per test whereas NVC takes around 5.6 s per test. The reason is the use of the large library of verification packages. Thus I wonder if there is any possibility of re-using intermediate LLVM-link results or any other mechanism to improve the situation? My feeling is that NVC will perform a lot of redundant work for each elaboration/run. 

Maybe this is something that has to be managed by an outside script, but in that case how?

nickg, 2015-06-02:
You can pre-compile packages to native code using the `--codegen` command. This is what happens with the IEEE libraries during bootstrap. In that case they won't be linked with `llvm-link` and won't be re-JIT-ed at simulation time (it just uses `dlopen`). E.g.

```
nvc --codegen my_pkg
```

I'd suggest doing that after you analyse the packages. 

We could still improve things, however, by not calling out to the external `llvm-link` and `opt` processes at all and doing it internally using the LLVM API.

kraigher, 2015-06-02:
Thank you this did improve the performance making NVC also run at about 0.2 s per test.
Is it only packages that can be pre-compiled or also entities/architectures? What about generic entities or VHDL-2008 generic packages (if you plan to support those).

For some people maybe it would be convenient if one could do the following to make NVC pre-compile everything it can in `file.vhd`:
```console
> nvc -a file.vhd --codegen
```
In VUnit we parse the code for every type of design unit so there I can add the extra codegen calls after compiling a file containing a package body.

nickg, 2015-06-14:
I've made a couple of improvements in this area:
* NVC no longer shells out to the external `llvm-link` and `opt` programs: instead it uses the API to perform these tasks correctly.
* It's now possible to chain commands together to avoid re-reading LLVM bitcode and library files between each step. For example `nvc -a foo.vhd -e foo -r` performs analysis, elaboration, and execution in a single command. The combination `-e foo -r` is equivalent to GHDL's `--elab-run`.

nickg, 2017-12-30:
This shouldn't be a problem anymore as we now always generate a native shared library for packages and elaborated designs. (So the `--native` flag doesn't actually do anything anymore and the JIT backend is only used when native compilation is disabled at configure time.)

closed 2017-12-30 by nickg