nickg/nvc #199: make check proc7 failed

postoroniy, 2015-06-02:
issue9 : ok
        record4 : ok
        record5 : ok
         const3 : ok
proc7 : failed
nvc  -a  nvc/test/regress/proc7.vhd
nvc  -e proc7 --disable-opt 
mv nvc/build/logs/proc7/work/_WORK.PROC7.elab.bc nvc/build/logs/proc7/work/_WORK.PROC7.final.bc
nvc/build/bin/nvc  -r proc7
** Note: 1ns+0: Report Note: '1'
	Process :proc7:line_24
	File nvc/test/regress/proc7.vhd, Line 16
nvc: ../src/object.h:113: tree_t tree_array_nth(tree_array_t *, unsigned int): Assertion `n < a->count' failed.
*** Caught signal 6 (SIGABRT) ***
-------- STACK TRACE --------
[0x40022d4c]
-----------------------------
...
FAIL: test/run_regr.rb
==============================================
1 of 13 tests failed

nickg, 2015-06-02:
I can reproduce this but only on a 32-bit system.

nickg, 2015-06-07:
Fixed now

closed 2015-06-07 by nickg