nickg/nvc #206: fatal: accessed 0 bytes beyond process temp stack region

kraigher, 2015-06-15:
```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity ent is
end entity;

architecture a of ent is
begin
  main : process
    variable u : unsigned(31 downto 0) := (others => '0');
  begin
    for i in 0 to 1023 loop
      report integer'image(i);
      u := u / 10;
    end loop;
    wait;
  end process;
end architecture;
```

```console
> nvc --messages=compact -a bug.vhd -e ent -r ent
...
bug.vhd:14:7: note: 0ms+0: Report Note: 53
	Process :ent:main
fatal: accessed 0 bytes beyond process temp stack region

-------- STACK TRACE --------
nvc(show_stacktrace+0x16) [0x416036]
nvc(fatal_trace+0x9e) [0x41610e]
nvc() [0x4162f4]
/lib/x86_64-linux-gnu/libc.so.6(+0x36d40) [0x7fc29c582d40]
/lib/x86_64-linux-gnu/libc.so.6(memset+0x2c) [0x7fc29c5d85ec]
/opt/nvc/share/nvc/ieee/_IEEE.NUMERIC_STD-body.so(IEEE.NUMERIC_STD.RESIZE$uIEEE.NUMERIC_STD.UNSIGNED;uIEEE.NUMERIC_STD.UNSIGNED;N+0x6d) [0x7fc29b3a41fd]
```

diogratia, 2015-06-17:
This on OS X.

````
** Note: 0ms+0: Report Note: 52
	Process :ent:main
	File bug_206.vhdl, Line 14
** Note: 0ms+0: Report Note: 53
	Process :ent:main
	File bug_206.vhdl, Line 14
*** Caught signal 10 (SIGBUS) [address=0x10b6e5018, ip=0x7fff8b218f6e] ***

-------- STACK TRACE --------  
1   libsystem_platform.dylib            0x00007fff8b218f6e _platform_bzero$VARIANT$Merom + 94  
2   ???                                 0x00007fff5520e42e 0x0 + 140734621606958  
3   _IEEE.NUMERIC_STD-body.so           0x000000010b5f497c IEEE.NUMERIC_STD.RESIZE__uIEEE.NUMERIC_STD.UNSIGNED__uIEEE.NUMERIC_STD.UNSIGNED__N + 108  
-----------------------------
````
I've always had this mental image of the expression stack being retired on statement boundaries. In this case the report and variable assignment statements.

The code fails when you increase the loop constant's right bound above 52.

Use an operand with a smaller expression stack footprint:
````VHDL
      -- u := u / 10;
      u := u srl 1;  
````  
And it finishes all 1024 iterations.

nickg, 2015-06-27:
This was broken by the switch to the new code generator.

closed 2015-06-27 by nickg