nickg/nvc #207: Memory corruption bug

forrestv, 2015-06-15:
I'm seeing a signal's value (specifically a value within a record within an array within an entity-level record) get corrupted in between it being written, the process being suspended, and it being read again. Any pointers on how to debug something like this?

postoroniy, 2015-06-16:
it'd be better to check and reproduce from source code rather than analyse your complain.

nickg, 2015-06-16:
I might be helpful to run the simulation with `--trace` and then grep out the interesting signals and post the output here?

nickg, 2015-07-06:
I just fixed something that might cause this behaviour. Can you test again? You'll need to do a `make clean` to rebuild the IEEE libraries.

nickg, 2015-07-23:
I think this was probably fixed by the above. Please reopen if not.

closed 2015-07-23 by nickg