nickg/nvc #210: Distinguish std_ulogic_vector and std_logic_vector in functions/aliases.

rhinton, 2015-07-18:
If you look closely at the std_logic_1164 package, you'll see that std_logic_vector (resolved) and std_ulogic_vector (unresolved) each are defined as an array type.  This important because VHDL's strong typing means that one won't be mistaken for the other -- kind of like types `SIGNED` and `UNSIGNED` in numeric_std.

However, when I try to compile code that defines functions for both cases, I get errors that the second function/alias is already defined.  In particular, I'm using the compatibility packages that offer many of the VHDL-2008 features in VHDL-93:

http://www.eda.org/fphdl/

When I compile `std_logic_1164_additions.vhdl`, I get the following errors:

    ** Error: WORK.STD_LOGIC_1164_ADDITIONS_C.TO_BV already declared in this region
            File /home/rhinton/projects/mvdlib/contrib/vhdl2008_compat/std_logic_1164_additions_c.vhdl, Line 24
          alias to_bv is ieee.std_logic_1164.To_bitvector [STD_ULOGIC_VECTOR, BIT retur ...
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ** Error: WORK.STD_LOGIC_1164_ADDITIONS_C.TO_BIT_VECTOR already declared
              in this region
            File /home/rhinton/projects/mvdlib/contrib/vhdl2008_compat/std_logic_1164_additions_c.vhdl, Line 26
          alias to_bit_vector is ieee.std_logic_1164.To_bitvector [STD_ULOGIC_VECTOR, B ...
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...

That first error is saying that the to_bv alias that takes `STD_ULOGIC_VECTOR` is conflicting with the one that takes `STD_LOGIC_VECTOR` (on the previous line in the VHDL source).

Any suggestions how to work around this?

I downloaded nvc and built from source on 18 Jul 2015.  I'm running CentOS 6.5 including LLVM 3.4.2 from the distro repo.  I'm happy to provide any further information about my system if you like, or to discuss the language issues if it helps.

Thanks for the great tool!

nickg, 2015-07-19:
This is basically the same thing as issue #89 so should be fixed by 8635622c as well. I've added a script to download and install the 2008 support library (`./tools/build-2008-support.rb`) and all the other files analyse OK now. I haven't tried running anything using it though.

closed 2015-07-19 by nickg

rhinton, 2015-08-05:
FYI, this works for me, too.  Thanks!