nickg/nvc #211: Add compatibility for older autoconf

rhinton, 2015-07-18:
I'm using autoconf 2.63 (CentOS 6.5), and it lacks the `m4_ifblank` and
`m4_ifnblank` macros.  Following a post from the octave list, I added the
following file in the distribution root:

acinclude.m4:
===
    ##rhinton: stolen from http://hg.savannah.gnu.org/hgweb/octave/rev/691545147aca
    dnl
    dnl If needed, define the m4_ifblank and m4_ifnblank macros from autoconf 2.64
    dnl This allows us to run with earlier Autoconfs as well.
    ifdef([m4_ifblank],[],[
    m4_define([m4_ifblank],
    [m4_if(m4_translit([[$1]],  [ ][        ][
    ]), [], [$2], [$3])])])
    dnl
    ifdef([m4_ifnblank],[],[
    m4_define([m4_ifnblank],
    [m4_if(m4_translit([[$1]],  [ ][        ][
    ]), [], [$3], [$2])])])
    dnl
===

After this addition, the remainder of your instructions worked beautifully.

nickg, 2015-07-19:
I put this in a separate file under `m4`. Can you check it still works?

rhinton, 2015-07-23:
Sorry for the slow response.  I'm traveling and don't have access to this machine.  I'll try it in a week or two when I get back.

rhinton, 2015-08-05:
OK, I'm finally back from vacation.  And your fix works perfectly: autogen.sh finishes without complaint.  Now I fail during configure; it appears my older autoconf doesn't recognize one of the macros, so it leaves it in.  I get the error 

```
../configure: line 9724: syntax error near unexpected token `-rdynamic,'
../configure: line 9724: `AX_CHECK_LINK_FLAG(-rdynamic,'
```

I see this macro called out in configure.ac line 68, and it sneaks through to the configure script at the line mentioned in the error.  For now I'm just commenting it out and continuing.  (Sorry I don't have a patch; I haven't been forced to learn autoconf yet.)

nickg, 2015-08-05:
I guess you have an older version of `autoconf-archive` without that macro. I added it to the `m4` directory in the repository.

closed 2015-08-05 by nickg

rhinton, 2015-08-06:
Confirmed the build completes perfectly now.  And I'm amazed at how quickly you close these bugs!  The fastest I've ever had one of our high-paid vendors release a bug fix is close to 2 weeks.