nickg/nvc #212: constant folding error during elaboration in altera_mf.vhd

michg, 2015-07-22:
Hi Nick,
I am getting an fatal error with altera_mf.vhd (Altera MF-Simulation library) at elaboration:
[image]

nickg, 2015-07-23:
Do you know where `intended_device_family` is defined? Is it a generic?

michg, 2015-07-24:
Yes, "intended_device_family" is a generic. Here is the definition:
[image]
[image]

nickg, 2015-07-25:
I've fixed that particular issue but unfortunately elaborating `altpll` still fails later on

michg, 2015-07-27:
Thanks for your update. I'm getting now the same message at this location:
[image]

nickg, 2017-05-05:
This works for me now, using the latest `master` branch. Albeit using a very old version of Quartus (13.0).

closed 2017-05-05 by nickg