nickg/nvc #218: Stack dump: conversion function on complex data type

rhinton, 2015-08-05:
I'm working on a conversion function for a fairly complex data type, and I get the following stack trace.

```
Name       WORK.DNS_QUERY_LIST_PKG.TO_SLV$VuWORK.DNS_QUERY_LIST_PKG.QUERY_LIST_RECORD;
Kind       function
Context    WORK.DNS_QUERY_LIST_PKG-body
Blocks     1
Registers  95
Types      27
  WORK.DNS_QUERY_LIST_PKG.QUERY_LIST_RECORD {0..8, [32] : 0..8, [16] : 0..8, [16] : 0..8, [64] : 0..8, [13] : 0..8, [13] : 0..8, [64] : 0..8, [128] : 0..8, [*] : 0..8, 0..8}
Variables  1
  RESULT                                // [*] : 0..8 
Result     [*] : 0..8
Parameters 1
  r0    QL_REC                          // @<WORK.DNS_QUERY_LIST_PKG.QUERY_LIST_RECORD{}> 
Begin
   0: r1 := const 0                     // -2^31..2^31-1 => 0 
      r2 := const 347                   // -2^31..2^31-1 => 347 
      r3 := const 4                     // -2^31..2^31-1 => 4 
      // Elided bounds check for r3 
      r4 := fcall WORK.MATH_MVD.CEILLOG2$II r3 // -2^31..2^31-1 
      r5 := add r2 + r4                 // -2^31..2^31-1 => -2147483301..2147483994 
      r6 := const 1                     // -2^31..2^31-1 => 1 
      r7 := add r5 + r6                 // -2^31..2^31-1 => -2147483300..2147483995 
      r8 := sub r7 - r6                 // -2^31..2^31-1 => -2147483301..2147483994 
      r9 := add r8 + r6                 // -2^31..2^31-1 => -2147483300..2147483995 
      r10 := cast r9                    // # => -2147483300..2147483995 
      r11 := const 0                    // # => 0 
      r12 := cmp r10 < r11
      r13 := select r12 then r11 else r10 // # 
      r14 := alloca r13                 // @<0..8> => 0..8 
      // Elided bounds check for r3 
      r15 := fcall WORK.MATH_MVD.CEILLOG2$II r3 // -2^31..2^31-1 
      r16 := add r2 + r15               // -2^31..2^31-1 => -2147483301..2147483994 
      r17 := add r16 + r6               // -2^31..2^31-1 => -2147483300..2147483995 
      r18 := sub r17 - r6               // -2^31..2^31-1 => -2147483301..2147483994 
      r19 := const 0                    // 0..1 => 0 
      r20 := wrap r14 [r1 r18 r19]      // [*] : 0..8 
      RESULT := store r20
      storage hint r14 length r13       // 0..8 
      // Begin dynamic aggregrate line 95 
      r21 := const 0                    // 0..8 => 0 
      // Elided bounds check for r3 
      r22 := fcall WORK.MATH_MVD.CEILLOG2$II r3 // -2^31..2^31-1 
      r23 := add r2 + r22               // -2^31..2^31-1 => -2147483301..2147483994 
      r24 := add r23 + r6               // -2^31..2^31-1 => -2147483300..2147483995 
      r25 := sub r24 - r6               // -2^31..2^31-1 => -2147483301..2147483994 
      // Elided bounds check for r3 
      r26 := fcall WORK.MATH_MVD.CEILLOG2$II r3 // -2^31..2^31-1 
      r27 := add r2 + r26               // -2^31..2^31-1 => -2147483301..2147483994 
      r28 := add r27 + r6               // -2^31..2^31-1 => -2147483300..2147483995 
      r29 := sub r28 - r6               // -2^31..2^31-1 => -2147483301..2147483994 
      r30 := add r29 + r6               // -2^31..2^31-1 => -2147483300..2147483995 
      r31 := cast r30                   // # => -2147483300..2147483995 
      r32 := cmp r31 < r11
      r33 := select r32 then r11 else r31 // # 
      r34 := alloca r33                 // @<0..8> => 0..8 
      r34 := memset r21 length r33
      r35 := wrap r34 [r1 r25 r19]      // [*] : 0..8 
      // Elided bounds check for r3 
      r36 := fcall WORK.MATH_MVD.CEILLOG2$II r3 // -2^31..2^31-1 
      r37 := add r2 + r36               // -2^31..2^31-1 => -2147483301..2147483994 
      r38 := add r37 + r6               // -2^31..2^31-1 => -2147483300..2147483995 
      r39 := sub r38 - r6               // -2^31..2^31-1 => -2147483301..2147483994 
      index check low r1 high r39 in 0..2^31-1 subkind 6
      // Elided bounds check for r3 
      r40 := fcall WORK.MATH_MVD.CEILLOG2$II r3 // -2^31..2^31-1 
      r41 := add r2 + r40               // -2^31..2^31-1 => -2147483301..2147483994 
      r42 := add r41 + r6               // -2^31..2^31-1 => -2147483300..2147483995 
      r43 := sub r42 - r6               // -2^31..2^31-1 => -2147483301..2147483994 
[... 96 lines cut ...]
              std_logic_vector(ql_rec.ns_host_next_ip) &
              ql_rec.ns_has_unresolved_hosts;
    return result;
  end function to_slv;
```

and the data type:

```vhdl
  type query_list_record is record
    valid                   : std_logic;  -- high if this record is valid
    origin_ip               : ip4_address;     -- request source IP address
    origin_port             : ip_port;  -- request source IP port
    dns_pkt_id              : uint16;   -- request DNS packet ID
    name_info                : dns_name_info; --std_logic_vector
    time_created            : ufixed(6-1 downto -7);  -- absolute time created in seconds
    last_activity           : ufixed(6-1 downto -7);  -- absolute last activity in seconds
    ns_name_info             : dns_name_info;  --std_logic_vector
    ns_host_ips             : ip4_address_vector(0 to LEN_HOST_IPS-1);  -- known IP addresses for current NS
    ns_host_next_ip         : unsigned(WI_HOST_IP_IDX-1 downto 0);  -- next NS IP to consult
    ns_has_unresolved_hosts : std_logic;  -- high when NS has unresolved hosts
  end record query_list_record;
```

Let me know the best way to proceed.

nickg, 2018-02-05:
Apologies for the lack of updates on this, could you try again with the latest master version? There have been a lot of improvements and fixes in this area.

nickg, 2018-02-13:
Going to close this for now, please reopen if it's still a problem.

closed 2018-02-13 by nickg

rhinton, 2018-02-26:
Sorry for the slow response.  Thanks for the update, I'll try again when convenient, though that may be a while.  I'm glad you've closed it for now.